christianhaitian/arkos #790: Updates

lauren7ino, 2023-09-15:
Hi Christian
One simples question:
When you release an Update, can I assume that this update brings everything up to date (on the date of the release), like RetroArch/32 cores, shaders, and PortMaster, etc?
Thanks

christianhaitian, 2023-09-16:
Not necessarily.  PortMaster updates directly when needed upon launching it.  Cores are updated on various schedules depending mostly when libretro has meaningful updates to the cores and you can update those directly from within retroarch and retroarch32.  Retroarch/Retroarch32 emulators are updated when there are new releases by libretro.  You can review the specific update changelog for your device here: https://github.com/christianhaitian/arkos/wiki/Changelogs

closed 2023-09-16 by christianhaitian

lauren7ino, 2023-09-16:
As I suspected. Thank you, keep up the good work!