christianhaitian/arkos #793: Loading mods that contain .bin files in ECWolf

BREAKHEART1, 2023-09-20:
### Describe the Issue (If applicable)
I am unable to load ECWolf mods that contain a .bin file (EG: https://www.moddb.com/mods/astrostein-for-ecwolf/downloads/astrostein-the-original-laz-rojas-scenario-wolf3d-14).

I have tried using the following .ECWolf files (Apologies if I am missing something obvious) and ECWolf crashes at the title screen and returns me to Emulationstation:

1. SUBDIR=WOLF3D
DATA=WL6 
PK3_1=/roms2/wolf/WOLF3D/astrostein.pk3
PK3_2=/roms2/wolf/WOLF3D/astrostein.bin

2. SUBDIR=WOLF3D
DATA=WL6 
PK3_1=/roms2/wolf/WOLF3D/astrostein.pk3
BIN=/roms2/wolf/WOLF3D/astrostein.bin

### How can the issue be reproduced? (If applicable)
See above. You can also replicate the issue on the PC version of ECWolf if you only drag and drop the .pk3 file onto the ECWolf executable. If you drag and drop both files onto the ECWolf executable, then the mod loads fine.

### What device are you using?
353VS

### What version of ArkOS are you on?
ARKOS 2.0 (08112023)

christianhaitian, 2023-09-20:
The 2nd option definitely wouldn't work as BIN is not a valid key.  In the download, the `.pk3` file is named `astrostein_fix_PC_v2.pk3`.  Did you rename that file to `astrostein.pk3` before attempting to launch this game?

BREAKHEART1, 2023-09-20:
Ah sorry, yes I renamed the files to Astrostein.pk3 and Astrostein.bin

christianhaitian, 2023-09-21:
Well, I was able to load Astrostein but it would crash as you try to start a game.  Here's the contents of my ecwolf file.

```
SUBDIR=Wolfenstein 3D
DATA=WL6
PK3_1=/roms2/wolf/astrostein_fix_PC_v2.pk3
PK3_2=/roms2/wolf/Astrostein.bin
```

BREAKHEART1, 2023-09-21:
Thanks Christian,

After starting from scratch and experimenting with the settings, I seem to have found a fix. I'll do my best to explain everything.

- I have my ROMs on a 2nd SD card
- The main Wolfenstein files are in /roms2/Wolf/WOLF3D
- The mod files are Astrostein.bin and astrostein_fix_PC_v2.pk3

I found that if I place Astrostein.bin and astrostein_fix_PC_v2.pk3 within the WOLF3D folder, then they will load fine using this ecwolf file:

SUBDIR=WOLF3D
PK3=astrostein_fix_PC_v2.pk3 Astrostein.bin
DATA=WL6

If I swap the positions of the PK3 and DATA lines, then the mod doesn't load:

SUBDIR=WOLF3D
DATA=WL6
PK3=astrostein_fix_PC_v2.pk3 Astrostein.bin

If I move the mod files into /roms2/Wolf/ and edit the .ecwolf file accordingly, then the mod only partially loads (The textures change, but the maps and title screen are original Wolfenstein 3D):

SUBDIR=WOLF3D
PK3=/roms2/wolf/sastrostein_fix_PC_v2.pk3 /roms2/wolf/Astrostein.bin
DATA=WL6

christianhaitian, 2023-09-21:
Interesting.  Thanks for documenting that here so others may benefit if needed in the future.

closed 2023-09-21 by christianhaitian