christianhaitian/arkos #794: Slower boot up time with large 'ports' folder

MetroidFusion, 2023-09-22:
Hi

I've discovered that when the 'ports' folder contains a lot of (game) files, the boot up time of ArkOS is significantly longer. I have tested this on numerous devices and also with the newest build (09212023). Without the 'ports' folder in the roms partition, the boot up time is around 14 seconds, but with the 'ports' folder (containing a lot of game files) it's approximately 25 seconds .

It seems like EmulationStation is scanning the entire 'ports' folder for files, instead of just the root of the folder where the .sh files are located.

Would it be change this assumed behavior, so that EmulationStation only scans the root of the 'ports' folder (for .sh files) to optimize boot up time?

/M

christianhaitian, 2023-09-22:
Go to the start menu>vram limit and try increasing the vram limit to 250Mb and see if that makes a difference.  Let me know your results.

MetroidFusion, 2023-09-22:
> Go to the start menu>vram limit and try increasing the vram limit to 250Mb and see if that makes a difference. Let me know your results.

I just tried increasing the vram limit to 250Mb (and higher), but is apparently did not make a difference.  When booting the device and EmulationStation is loading the different platforms, it kind of 'hangs' for a few seconds when the (large) ports folder is available in the roms partition. It  hangs on either 'Uzebox' or 'Movies', even though I don't have any files in these folders. When the ports folder is not available in the roms partition, the device boots significantly faster.

As mentioned, it seems like EmulationStation is scanning the entire 'ports' folder for files, instead of just the root of the folder where the .sh files are located.

christianhaitian, 2023-09-23:
Hi,

See if this helps with this issue on your end.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and see if it speeds up the load time for ES.

Let me know your results.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12705945/arkosbackup.tar.gz)

MetroidFusion, 2023-09-23:
Wow, that certainly resolved the issue! It boot just as fast with or without the 'ports' folder now. I wondered why the boot up time became slower over the years of using ArkOS on different devices, but It might just be due to the fact, that I've during this period have added more and more ports (and therefore a large amount of game files) to my setup. But now it's totally fixed. Thank you so much! :)

christianhaitian, 2023-09-23:
Awesome!  Thanks for confirming.  I'll queue this fix for the next update for others.

closed 2023-09-23 by christianhaitian