christianhaitian/arkos #797: rg353v never gets pass the initial first boot

tallpr24, 2023-09-23:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

tallpr24, 2023-09-23:
just received my rg353v, gets pass the arkos screen, then i get a black screen. tried using 3 different sd cards, tried using rufus and win32 disk imager, and tried another computer, but Jelos works with the same sd cards.

any suggestions?

christianhaitian, 2023-09-23:
Are you making sure not to format the linux partition after the flash has completed?  Also made sure to use a Sandisk or Samsung sd card?

lpbousquetl, 2023-09-24:
Hi, similar problem for my, got a brand new rg353v  from Anbernic and I can't get past the ArkOS boot screen.  Using samsung Evo 64GB card  Tried to flash ArkOS_RG353V_v2.0_05042023.img.xz  with Win32DiskImager, rufus and usbit.

lpbousquetl, 2023-09-24:
Finally I got someting.
I rebooted multiple time while pressing B and I finally see the install script running then it hang. So I rebooted with B pressed and I finally saw a menu where I choose emu station(First choice.).  I rebooted the system with the reset button and it came back. But I tried to shutdown the system then boot it again and it's now stuck on the ArkOS boot logo.

christianhaitian, 2023-09-25:
Must be something with newer units is my guess as I'm able to boot from a fresh flash on samsung, sandisk, and kioxia cards I have on my 353v1 and 353v2 units.

christianhaitian, 2023-09-25:
I disabled the placebo overclock setting in the rk3566-OC.dtb file located on the BOOT fat32 partition.  You can try copying the one in the attached zip file into that BOOT folder overwriting the existing and see if that resolves the issue.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/12710876/rk3566-OC.zip)

tallpr24, 2023-09-25:
After , 4 different cards and 3 different computers I gave up, jelos worked 100% on the same cards.

closed 2023-09-27 by christianhaitian