christianhaitian/arkos #798: Quick Mode not functioning properly after update (09232023)

MetroidFusion, 2023-09-24:
Hi

Just wanted to report that Quick Mode seems to not work as intended after the latest update (09232023). If you boot up a game and turn the device off, it will save the state and reload it upon reboot. But if you then try to turn off the device directly after this sequence (again, using the Quick Mode feature), the screen will simply go blank/dark. You then have to press the power button yet again for the device to actually turn off. It does still save the state and reload it upon reboot. This was not the behavior before this update. I have tested this on both my primary setup, and on a fresh install with the same result.

/M

christianhaitian, 2023-09-24:
I was able to replicate this.  I'm investigating it.

christianhaitian, 2023-09-24:
Odd, it's not doing it again.  It's like after the update, this occurs after the first attempt to shutdown from quickboot but after that, it doesn't occur again on my end.  Can you confirm this on your end or are you able to reproduce this issue consistently?  If so, what game and emulator have you tried?

MetroidFusion, 2023-09-24:
I'm able to reproduce this issue consistently. I have tried different games and emulators (mGBA, Gambatte etc.) with the exact same result; the screen will simply go blank when pressing the power button after using Quick Mode prior to this step, and I will have to press it again for the device to actually turn off. As mentioned, Quick Mode did not exhibit this behavior before the latest update.

christianhaitian, 2023-09-24:
Did you enable the tap to power off the device in emulationstation so you don't have to do F+Power to turn off?

MetroidFusion, 2023-09-24:
I have actually. But I have also tried deactivating this feature with the same result. And just to be sure, I have also flashed the OS to another SD card, without messing with settings etc. afterwards. But the Quick Mode issue persists.

christianhaitian, 2023-09-24:
Ok, try restoring this backup that contains the updated scripts for quickmode and see if your results are any different.  Make sure you have quickmode enabled in emulationstation first before restoring.  If you don't have it enabled before restoring this backup, just do a Disable Quick Mode then Enable Quick Mode before you start testing.  Let me know the results.  Thanks.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12708956/arkosbackup.tar.gz)

christianhaitian, 2023-09-24:
I found the issue and will have another update out shortly.

MetroidFusion, 2023-09-24:
Awesome, that sounds great! I'm looking forward to the update! :)

christianhaitian, 2023-09-24:
An update is now available that should fix this quickmode issue.  Please update and let me know your results.

MetroidFusion, 2023-09-24:
I have now updated and tested the update. The issue with the blank screen is now fixed. However, I noticed that when you exit a game that has been suspended in Quick Mode, this game does not show up under the 'Last Played' section. In version 09212023 games that were suspended using Quick Mode did show up under this section. I remember when using Quick Mode in this version, you could see that the script exited out of EmulationStation (very briefly), and then turned the device off (this process probably 'registered' the game under the 'Last Played' section). Now, in the newest update, the device turns off immediately, and the game is not registered. But your fix does indeed fix the blank screen issue.

christianhaitian, 2023-09-25:
The game not showing up in Last Played is a known limitation.  I don't have a solution to address that yet.  surprising that it worked previously as that never did for me.  Thanks for confirming that this update fixed this particular issue.

closed 2023-09-25 by christianhaitian

MetroidFusion, 2023-09-25:
I was surprised as well. But every game was consistently registered under the 'Last Played' section when using Quick Mode in 09212023, which was quite nice. Stability is of course the most important thing, but It would be pretty awesome if this 'feature' could be implemented in a future build. I wonder if games could be 'registered' when launching them, instead of when exiting them. Just a thought.

MetroidFusion, 2023-09-25:
Okay, so I just figured out how to make games show up under the 'Last Played' section when using Quick Mode'. You simply have to disable the launch image (i.e. rename/delete 'loading.jpg in /roms/launchimages). By doing that every game will consistently be registered under the 'Last Played' section when using Quick Mode.

(I was initially testing the newest update on a fresh install where the launch image was not disabled. This certainly explains why I observed that games were consistently registered under the 'Last Played' section in previous versions using Quick Mode, as I prefer not to have a launch image when booting up a game. So your newest Quick Mode script is superior in every way compared to earlier versions; It's quicker, more reliable, and retains the ability to register games under 'Last Played' (if you disable the launch image).

MetroidFusion, 2023-09-25:
Upon further testing I have discovered that the newest version of Quick Mode does not work reliably with PS1/PSX games (using retroarch32/PCSX ReARMed). More often than not I receive the message 'Failed to load state from (insert game title, e.g. Final Fantasy VII)'.

christianhaitian, 2023-09-25:
The first version didn't work reliably with ps1 games either.  Save states failed more consistently with the first version than this one.  I believe the issue is that the process doesn't wait long enough for the auto savestates to complete before shutting down for cd based systems.   Maybe not all, but certainly for psx and dreamcast.   I haven't figured out a reliable solution for those yet.  I did find a solution for the last played collections though.

christianhaitian, 2023-09-25:
Now I think I have a solution for quick mode that will allow the script to wait until retroarch has safely shutdown before powering down the system.  Please restore this backup and help me test and see if it improves reliability of auto save states for ps1.  Be aware that if you test dreamcast, the flycast32_rumble core has issues with auto savestates even outside of quick mode.  Regular retroarch/flycast is more reliable with auto savestates.

Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12711189/arkosbackup.tar.gz)

MetroidFusion, 2023-09-25:
Sounds really nice that you found a solution for the last played collections - it would be awesome to have an implementation that works whether or not you have a launch image enabled.

I have just tested the backup file, and I does not seem to fix the issue on my end when it comes to PS1 games. I still experience the same 'Failed to load state' message. As far as I remember Quick Mode did actually work for this system on update 09212023.

christianhaitian, 2023-09-25:
Sorry about that, the backup I provided didn't have the right files in it.  Try this one instead.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12715998/arkosbackup.tar.gz)

MetroidFusion, 2023-09-25:
I have been 'stress testing' your newest backup/solution for a few hours, and It seems like it solves the issue with PS1 games in particular. That's awesome, I believe Quick Mode works as intended now! :)

christianhaitian, 2023-10-13:
Hi, is quick mode still working well for you?  I've got a person who's reporting that when they attempt to shutdown the device a second time after a game has been resumed in quick mode, the device just goes straight to emulationstation.  I can't replicate this condition on my end.