christianhaitian/arkos #799: OGA1.1 retroarch unable to load after update

Shally1986, 2023-09-24:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes 
3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? Yes 

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? Yes 

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Unable to launch any game which uses retroarch or retroarch32 . Unable to launch retroarch from the menu . Tried to reset the settings . Still unable to launch . 
### How can the issue be reproduced? (If applicable)
Launch any game using retroarch . It does not launch . I get the arkos splash screen twice and return back to launch screen . 

### What device are you using?

OGA1.1

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
09232023

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-09-24:
I can't replicate this issue on my end.  You can try to restore this file and see if that resolves the issue for you.  It's only good for the OGA, RK2020, and RGB10 devices only.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/12709052/arkosbackup.tar.gz)

Shally1986, 2023-09-24:
Thank you !! It works . Any idea what might have caused ? Apart from updating arkos and turning on quick mode , I didn't make any changes

christianhaitian, 2023-09-24:
No idea as I can't replicate the issue on my end.  Happy it worked though.

closed 2023-09-24 by christianhaitian