christianhaitian/arkos #800: TMNT Port broke since latest September update

evillurker, 2023-09-26:
Device - RG353V

The latest update that works on this port was from July, after that - updating to the september update something broke the port and its no longer working. 
Any chance to fix it?

Log file:

assemblyName: TMNT
appBasePath: /roms2/ports/tmntsr/gamedata
appArgs: {}
Redirecting ParisEngine to MONOMODDED_ParisEngine.dll
System.IO.FileNotFoundException: Could not find file "/home/ark/.local/share/Tribute Games/TMNT/ModSettings.dat"
File name: '/home/ark/.local/share/Tribute Games/TMNT/ModSettings.dat'
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x001ef] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode) [0x00000] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode)
  at Paris.Game.Menu.Options.LoadModConfig () [0x00006] in <b040a8aeb1a640ccbfa8073dae206c9c>:0 
Params Set:
> None
Removing Program.CrashHandler handler...
Controller 0: retrogame_joypad, GUID: 4b480111, Mapping: 19009b4d4b4800000111000000010000,retrogame_joypad,b:b1,a:b0,back:b8,dpdown:b14,dpleft:b15,dpright:b16,dpup:b13,leftshoulder:b4,leftstick:b11,lefttrigger:b6,lefty:a1,leftx:a0,rightshoulder:b5,rightstick:b12,righttrigger:b7,righty:a2,rightx:a3,start:b9,y:b2,x:b3,platform:Linux, 
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.gpath.c:115: assertion 'filename != NULL' failed

gpath.c:115: assertion 'filename != NULL' failed

FNA3D Driver: OpenGL
OpenGL Renderer: Mali-G52
OpenGL Driver: OpenGL ES 3.2 v1.g2p0-01eac0.02b077cea4dc2127696dc49c93c0fc7a
OpenGL Vendor: ARM
MojoShader Profile: glsles
Attempting to patch...

=================================================================
	Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

=================================================================
	Native stacktrace:
=================================================================
	0x558a6df9d8 - mono : 
	0x558a6dfce4 - mono : 
	0x558a697b18 - mono : 
	0x558a660a08 - mono : 
	0x7f9c5ff5ec - linux-vdso.so.1 : __kernel_rt_sigreturn
	0x7f9c549578 - /lib/aarch64-linux-gnu/libpthread.so.0 : raise
	0x7f9264f56c - /lib/aarch64-linux-gnu/libSDL2-2.0.so.0 : 
	0x7f9264f61c - /lib/aarch64-linux-gnu/libSDL2-2.0.so.0 : 
	0x7f9c5ff5ec - linux-vdso.so.1 : __kernel_rt_sigreturn
	0x7f8809bfcc - Unknown

=================================================================
	Telemetry Dumper:
=================================================================
Pkilling 0x547760460160x from 0x548084352064x
Pkilling 0x547758358912x from 0x548084352064x
Pkilling 0x547762561408x from 0x548084352064x
Pkilling 0x547879170432x from 0x548084352064x
Pkilling 0x548030390656x from 0x548084352064x
Entering thread summarizer pause from 0x548084352064x
Finished thread summarizer pause from 0x548084352064x.
Failed to create breadcrumb file (null)/crash_hash_0x636cd7e16

Waiting for dumping threads to resume

=================================================================
	External Debugger Dump:
=================================================================
mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb

=================================================================
	Basic Fault Address Reporting
=================================================================
Memory around native instruction pointer (0x7f9c549578):0x7f9c549568  02 00 80 d2 03 01 80 d2 e8 10 80 d2 01 00 00 d4  ................
0x7f9c549578  a5 d8 47 f9 e1 97 40 f9 a0 00 40 f9 20 00 00 ca  ..G...@...@. ...
0x7f9c549588  80 01 00 b5 e0 03 04 2a f3 0b 40 f9 fd 7b d3 a8  .......*..@..{..
0x7f9c549598  c0 03 5f d6 a1 00 00 d0 21 c4 47 f9 42 d0 3b d5  .._.....!.G.B.;.

=================================================================
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at System.AppDomain:ExecuteAssembly <0x00007>
	  at System.AppDomain:ExecuteAssemblyInternal <0x00047>
	  at System.AppDomain:ExecuteAssembly <0x0002f>
	  at System.AppDomain:ExecuteAssembly <0x00093>
	  at System.AppDomain:ExecuteAssembly <0x001db>
	  at System.AppDomain:ExecuteAssembly <0x0010f>
	  at System.AppDomain:ExecuteAssembly <0x00063>
	  at MMLoader:Main <0x005fb>
	  at <Module>:runtime_invoke_void_object <0x000eb>
=================================================================

christianhaitian, 2023-09-26:
See the pinned issue [here](https://github.com/christianhaitian/arkos/issues/780) about how this can be worked around until the porter can work on addressing updates needed in the port.  Note that the port is still considered beta and may break again with future OS updates.

closed 2023-09-26 by christianhaitian

evillurker, 2023-09-26:
Nevermind I found a solution, it seems that whoever edited the initial controls.txt file will have a problem since it was updated. Solution is to just use the default control.txt file. Thanks for your response!