christianhaitian/arkos #801: Screen is rotated after update to 09242023 version

wojtekjaskula, 2023-09-28:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)

After Updating R35S handheld to most recent arkos (09242023) screen is rotated and there is no option to change screen rotation settings except enabling developer mode (found on reddit) 

### How can the issue be reproduced? (If applicable)

Just update to the most recent version via Settings->Update

### What device are you using?

R35S handheld 

### What version of ArkOS are you on?

09242023

### Anything else you'd like to include that may help to review this issue or feature request?

I can provide photos , logs , anything which is needed to solve this issue.

christianhaitian, 2023-09-28:
See here for a possible fix. < https://github.com/christianhaitian/arkos/issues/791#issuecomment-1722408064>

I don’t have the device in hand just yet but once I do, I will provide a specific image for this device IF, I mean **IF** I decide to officially support it.  This is a **BIG IF**.

wojtekjaskula, 2023-09-28:
Thanks for answer @christianhaitian ! 

Is there any option to disable developer mode ? I don't see it currently under any settings

christianhaitian, 2023-09-28:
No.  Once it's enabled, you're only option is to reimage.

christianhaitian, 2023-09-28:
Too fix the rotation though, you can also try doing the following to fix the rotation,

1. SSH into the device.  See [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-do-i-ssh-into-ArkOS) for instructions on how to do this if need be.
2. Paste the following command in it. `sudo ln -sfv /usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.2600.2 /usr/lib/aarch64-linux-gnu/libSDL2.so && sudo ln -sfv /usr/lib/arm-linux-gnueabihf/libSDL2-2.0.so.0.2600.2 /usr/lib/arm-linux-gnueabihf/libSDL2.so`
3. Press Enter.
4. Reboot your device.

The creator of this device should have disabled the update feature on this device or conferred with me on how best to handle support if they were planning to use the OS and expect that it could be properly be updated from the git.

nunits, 2023-09-29:
> The creator of this device should have disabled the update feature on this device or conferred with me on how best to handle support if they were planning to use the OS and expect that it could be properly be updated from the git.
All of us who wanted to update have gone through the hell of the 90 degree screen. I solved it by formatting and installing your version ArkOS_RG351MP_v2.0_03112023.img.xz With that and copying the new rk3326-rg351mp-linux.dtb to the boot partition. Everything was already excellent. And it also allows updating without problem.
Hopefully we can have an exclusive version for the r35s.

christianhaitian, 2023-09-29:
Hopefully, once I have the device, I can figure out how to have the stock os update properly without needing a complete reflash if I decide to support the unit.

christianhaitian, 2023-10-07:
This issue has been addressed with today's update.

closed 2023-10-07 by christianhaitian