christianhaitian/arkos #804: Missing boot partition on RG351MP (Updated 03/11/2023)

GeGerardo, 2023-09-30:
### Describe the Issue
After "flash" the image in a microsd I can not find the Boot partition. I tried on windows, linux (Ubuntu 22) and macOS.

### How can the issue be reproduced? (If applicable)
1. Flash the image with DD, rufus or Balena
1. Try to mount the partitions again
1. Just EASYROMS partition is visible

### What device are you using?
I tried with a Sandisk 32GB and 64GB, Samsung Evo 64GB and an old Verbatim 16GB

### What version of ArkOS are you on?
RG351MP (Updated 03/11/2023)

### Anything else you'd like to include that may help to review this issue or feature request?
I need to have access to the boot partition do change the dtb binary

GeGerardo, 2023-09-30:
I managed to show it on Windows using WSL2 with ubuntu. That is so strange

christianhaitian, 2023-09-30:
Ahh yes, I need to update that image to correct that issue that some encounter

christianhaitian, 2023-09-30:
Updated image is now available that should resolve that issue for most if not all.  Thanks for reporting this.

closed 2023-09-30 by christianhaitian