christianhaitian/arkos #807: Wifi problem OGA 1.1/Updates

Elmagob99, 2023-10-05:
Hi everyone, im having issues to connect to wifi,my console has a wifi chip and i even tried with a dongle,i cant,the wifi symbol wont show up on the home screen , i also dont get a confirmation of correct network connection.There is some way to update the OS manually moving some files from the pc? i have the 08/11/2023 update and i just checked and im 3 updates behind,also i just tried and i cant even get pass the wifi menu, i press the + symbol and gets freezed.

christianhaitian, 2023-10-05:
Did you follow these instructions for configuring your wifi? <https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RGB10#q-how-do-i-configure-my-wifi-connection-in-arkos>

Elmagob99, 2023-10-05:
> Did you follow these instructions for configuring your wifi? https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RGB10#q-how-do-i-configure-my-wifi-connection-in-arkos
> 
> You press the R1 button to scroll through the wifi menu. The other options is that you can configure it through retroarch and use a text file. That's all explained in the instructions provided there.

i will try the text thing, just in case im just doing a backup of roms,the image on the wiki is updated? because if i cant connect i just will accumulate some updates and reflash the card every once in a while

Elmagob99, 2023-10-05:
> The link on the wiki is the latest image available. All updates are available through OTA updates until an updated image is made available which I probably won't provide until later this year.

i will update the results of the text file when i try it, thanks a lot!

Elmagob99, 2023-10-05:
Update! the wifikeyfile.txt worked perfectly, im alredy up to date, thanks christian

closed 2023-10-05 by Elmagob99