christianhaitian/arkos #808: ThemeMaster Fails to Detect Connection

spirit-in-exile, 2023-10-07:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
- A clear and concise description of what the bug is.

After updating ArkOS from v2.0 09292023 to 10062023, ThemeMaster tool (both v4.4.2 and v4.5.0) thinks device is offline. 09292023 with ThemeMaster v4.4.2 worked when last tried, (approx. 3 days ago)?

### How can the issue be reproduced? (If applicable)

Updated ArkOS from v2.0 09292023 to 10062023 using Options menu Update option.

After successful update completion, attempted to run ThemeMaster v4.4.2, and it thinks I’m offline.

Checked WiFi utility, network connection shows as present and active.

Checked JohnIrvine1433/ThemeMaster and  Jetup13/Emulationstation-OGA-Theme-Gallery repos, both active and otherwise accessible online. 

Deleted and recreated network connection, restarted device from ES Quit menu, verified connection present and active in WiFi utility, tried ThemeMaster again, still thinks it’s offline.

Cycled WiFiOff/WiFiOn in Options/Advanced, with and without restarting via ES Quit menu, with and without verifying connection present and active in WiFi utility, tried ThemeMaster again, same result.

Cycled Enable/Disable Remote Services in Options, with and without restarts between changes, no change.

Tested other ArkOS online functionality (scraper, PortMaster), connections were successful with both, but still not ThemeMaster.

Found on JohnIrvine1433/ThemeMaster repo that ThemeMaster had been updated to v4.5.0. Manually updated ThemeMaster on device via SSH and retried. ThemeMaster Startup now reflects version ThemeMaster v4.5.0, but even after all on/off disable/enable toggles and restart repeated, ThemeMaster is still not detecting connection.

### What device are you using?

Anbernic RG353M, SanDisk Extreme cards for OS and SD2, 49% charge.

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

v2.0 (10062023) 

### Anything else you'd like to include that may help to review this issue or feature request?

[image]

Will also report to JohnIrvine1433/ThemeMaster if possible/applicable.

christianhaitian, 2023-10-07:
ThemeMaster 4.30, 4.4.2 and 4.5.0 detect a connection just fine on the latest ArkOS 2.0 10062023 update for me.  I can't replicate this issue yet.  Tried on my rg353m, rg503, and rgb30 units.  All worked fine.

christianhaitian, 2023-10-07:
ThemeMaster will think you're offline if it can't ping google.com.  Are you able to ping google.com from where you're located?

spirit-in-exile, 2023-10-07:
> ThemeMaster 4.30, 4.4.2 and 4.5.0 detect a connection just fine on the latest ArkOS 2.0 10062023 update for me. I can't replicate this issue yet. Tried on my rg353m, rg503, and rgb30 units. All worked fine.

Perhaps it is my network then. It’s the only thing I’m presently unable to try differently. If it’s working for you, then it’s clearly not ArkOS or ThemeMaster. Thanks for checking!

spirit-in-exile, 2023-10-07:
> ThemeMaster will think you're offline if it can't ping google.com. Are you able to ping google.com from where you're located?

I believe so, no problems checking Gmail or YouTube, at least. I will have to try a different network when I am able.

closed 2023-10-07 by spirit-in-exile

spirit-in-exile, 2023-10-07:
> ThemeMaster will think you're offline if it can't ping google.com. Are you able to ping google.com from where you're located?

Just to follow up: I dug around in the thememaster files and changed the url for the test ping to another server, and now it’s working. Guess I wasn’t able to ping Google after all. Thanks for the save!