christianhaitian/arkos #810: exFAT EASYROMS partition still impossible to mount on the CHI

JayBee-git, 2023-10-07:
### Describe the Issue (If applicable)
It's still impossible to mount the EASYROMS in windows, it's not recognized as a readable partition by the disk manager.
Using the exFAT fix on ArkOS menu doesn't work (it doesn't detect any issue).
Note that it's properly recognized before the resizing, but that operation seems to corrupt the partition table.
I tested several SD cards of various sizes, mostly 32GB and more.

### How can the issue be reproduced? (If applicable)
Just flashing the image clean to a SD card, boot it on the CHI for the first initialization, then plug it back to windows.

### What device are you using?
GameForce CHI

### What version of ArkOS are you on?
v2.0 (09/24/20230)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-10-07:
Yea I need to update the image for the chi to resolve it but I no longer have a Chi so I don't know when I will be able to resolve that for the Chi.  The partition is not corrupted, just not set in manner that some computers can easily see it.  For now, what you can do is when you burn the image on to the sd card, before inserting the sd card inti the Chi, replace the `expandtoexfat.sh` file in the boot partition with this one https://github.com/christianhaitian/rk3326_core_builds/raw/rk3326/shell-scripts/expandtoexfat.sh then insert the sd card into the unit.

If you already have a current setup you'd prefer to not have to recreate, If you're comfortable with ssh, you can remote into the unit and run this command `printf "t\n3\n7\nw\n" | sudo fdisk /dev/mmcblk0` and that should take care of the issue as well.  It should be non destructive but I can't guarantee that due to the nature of this hobby.

JayBee-git, 2023-10-07:
Thanks a lot for the quick reply, I will try it out!

JayBee-git, 2023-10-07:
The SSH command worked perfectly, it's now back to normal exFAT. Thanks for the help!

christianhaitian, 2023-10-07:
Awesome.  Enjoy

closed 2023-10-07 by christianhaitian