christianhaitian/arkos #811: ArkOS frequent reboots just to get to the dash, exact issue as issue #504

tallpr24, 2023-10-13:
after buying several samsung and sandisk cards, the issue remains, arkos boots to the arkos screen, just sits there.  i have to reboot several times sometimes up to 8 times before it loads emulation station. i suspect it has something to do with the sdcard reader driver.

this is rg353v.  Jelos and stock work perfect

after it finally booted i was able to update to the latest version 10062023

but still problem persists.

even if i take the second sd card out , the one with the roms, the problem is the same.

christianhaitian, 2023-10-13:
How did you image the card?

christianhaitian, 2023-10-15:
Any update on my question?

tallpr24, 2023-10-15:
Apologies for the delay.  I first wipe all partitions from the card, then I formatted the card at exfat. The i tried using Rufus, win32 disk imager. I also tried balena.  Yesterday I received my second rg353v( my 10 year old wanted one) and I'm having the same issues.

christianhaitian, 2023-10-15:
Don't use Balena Etcher.  Try this tool and see if you have any better results. 
https://www.alexpage.de/usb-image-tool/download/

Also be sure you're using the updated image I uploaded yesterday.

tallpr24, 2023-10-15:
Are there any logs we can look at to see where it's failing? Also when I'm re imaging the card some times it takes 2 or 3 reboots before the process can start, same issue on both rg353v and different sd cards, but both systems work fine with jelos and stock

christianhaitian, 2023-10-15:
Only when you get to the OS.  From what you're describing, the unit doesn't get to the OS.

tallpr24, 2023-10-15:
It gets to the os after several reboots

tallpr24, 2023-10-15:
i cant find a donwload link for the new version you menitoned you posted yesterday. can you post the link?

christianhaitian, 2023-10-15:
They are located here: https://github.com/christianhaitian/arkos/wiki#download-links

tallpr24, 2023-10-15:
got it, downloading now, will try that along with our imaging app, will report back soon.

than you  :)

tallpr24, 2023-10-15:
tried it, its now stuck on the arkos screen

tallpr24, 2023-10-15:
rebooted, 3rd time, it now says arkos 2.0 (10062023)
umount: /roms: not mounted, just stuck there.  there is no sd card on the second slot

tallpr24, 2023-10-15:
rebooted again, now its gone further to the bluscreen with white window easyroms partition expansion, seems to be stuck again.
rebooted again, not its continuing the process,

christianhaitian, 2023-10-15:
At that point, it's in the OS.  It's just not able to continue for some reason without issues.  Have you fully charged the device?

tallpr24, 2023-10-15:
ran thru all the white letters, fast scrolling, rebooted itself, stuck on arkos screen again,, rebooted it, messaga about arkos with little dots, then black screen, froze at black screen. rebooted, arkso 2.0 (10062023) blinking dots, stuck with blinking dots

tallpr24, 2023-10-15:
I left it at the blinking dots, but then after a few minutes, i got a black screen.  just stuck at the black screen

christianhaitian, 2023-10-15:
Have you charged the device?

tallpr24, 2023-10-15:
rebooted again, now it loaded arkos, im at the dash, shutdown, turned on again, emulation station loading, black screen again. the device is at 92% battery

tallpr24, 2023-10-15:
rebooted again, dash loaded again.

tallpr24, 2023-10-15:
maybe we can look at logs since it loaded?

tallpr24, 2023-10-15:
just to test, i hit quit , restart, this time it loaded emulation station, shutdown to test, turned on again , stuck on arkos screen again, its inconsitent, pretty much same simptoms as before.

christianhaitian, 2023-10-15:
Are you sure you sd cards are authentic Samsung or Sandisk cards?  I don't experience any of this on my 3 rg353v devices.

tallpr24, 2023-10-15:
yes, i have 2 sandisk and 1 samsung, i able to reflash stock and jelos on both and they all work perfectly, and im having the same issue on 2 different rg353v consoles. i just restarted again, and its stuck on arkos again. could it be something with the card reader driver?

christianhaitian, 2023-10-15:
same driver used in JELOS.  again, I use the same image on my 3 rg353v without these issues and there are others who do as well.  Just some devices are compatible with ArkOS and some are not for one reason or another.  Although, I wonder if disabling the fake overclock may help in your case since you're able to boot sometimes.

christianhaitian, 2023-10-15:
Another difference, both JELOS and the Stock OS don't enable rumble support.  Perhaps that could be a source of instability for some units as well.  Going to try disabling that and the fake overclock give you an updated dtb to place in the fat32 BOOT partition and see if t makes a difference for stability for you

christianhaitian, 2023-10-15:
Try replacing the existing rk3566-OC.dtb file in the fat32 BOOT partition with the one size the attached zip file.  You can do this with the sd card inserted in your computer.  Then try booting ArkOS with the sd card and see if it makes a difference with stability.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/12910209/rk3566-OC.zip)

tallpr24, 2023-10-15:
Will try that by the way I rebooted again since it froze after I clicked on Wi-Fi, I rebooted and it got stick on the blinking dots , I click volume down and it looks like it's doing something but it's stuck on some kind of loop. Will try to upload a photo now

tallpr24, 2023-10-15:
[image]

tallpr24, 2023-10-15:
If I click the down volume again it brings me back to this [image]

tallpr24, 2023-10-15:
now its freezing up on the dots, :(

tallpr24, 2023-10-15:
going to re image the sandisk, that was the samsung, will report back

tallpr24, 2023-10-15:
[image]

tallpr24, 2023-10-15:
[image]

tallpr24, 2023-10-15:
Froze there. This is using the re images Sandisk card

tallpr24, 2023-10-15:
Going to head to Best Buy to buy another sd card

tallpr24, 2023-10-15:
Rebooted and now have this message [image]

tallpr24, 2023-10-15:
Looks like it's failing something during boot[image] froze here

tallpr24, 2023-10-15:
That's with the file you gave me

christianhaitian, 2023-10-15:
Well, I don't know unfortunately.  Another idea is to try a non quality sd card see if that acts differently.

christianhaitian, 2023-10-15:
Interesting, I just saw a report in the RGH Discord where someone was having a similar issue as you.  Stock OS worked fine but ArkOS was having mounting issues and UnofficialOS (old fork of JELOS).  The person decided to order replacement unit through Amazon and that unit works just fine with both OS.

tallpr24, 2023-10-15:
ok, so picked up a new sandisk from BestBuy, same steps but same results.  i guess these 2 units have something different that make it not work with arkos, i got mine china seller.  :( ill just stick to stock os for now i guess.
thanks for all your help. your OS looks amazing, but i have not tried it yet, maybe one day I will lol

tallpr24, 2023-10-15:
not sure why, mut jelos latest version runs perfect, and so does stock.

tallpr24, 2023-10-15:
would you mind sending me an older version of your os?  i read on reddit , another person had the same issue, but when he used 1.7 it worked perfect.  I just want to test to see whats going on

christianhaitian, 2023-10-15:
There was never a 1.7 for this unit.  1.7 existed on rk3326 devices last on the rg351p.  Be careful what you read on reddit.  There's a lot of misinformation on there.  Thanks for your time and sorry we couldn't figure this out.

closed 2023-10-15 by christianhaitian

tallpr24, 2023-10-15:
do you have another version older? id like to give it a try if you dont mind, the first version for the rk3326? and no worries, I apreciate your time

christianhaitian, 2023-10-15:
That will not work on this device.  The oldest version I have is the previous version you already tried.  Sorry.

tallpr24, 2023-10-15:
ok, :(