christianhaitian/arkos #812: HELP! R35S Arkos Stock is very large it cannot be installed like the rest of Arkos

estherliquid, 2023-10-13:
The Arkos Stock image of the R35S occupies 30 gigabytes and the console has a 16 GB microSD. How do I Flash the image?  It tells me that it doesn't have space, I use 2 cards, one for the system and the other for the ROMs. I don't know why that image takes up so much space.

The rest of Arkos's images take up a few gigabytes. Is it normal that the other one takes up so much?  How can I restore my console?  I hope you can help me, Thanks.

christianhaitian, 2023-10-13:
That's how the stock image came from the manufacturer. You'd have to ask the manufacturer about that.  If that's a problem for you, you can use the rg351mp image and update that to the latest ota update.  Good luck.

closed 2023-10-13 by christianhaitian

DynaMight1124, 2023-10-23:
Some R35S consoles came with 32GB cards and others came with 16GB. The stock image you're referring to was probably taken from a 32GB card, it wont have 32GB of data on it but its a clone of the whole card. There are some 16GB images floating about. Heres one: https://mega.nz/file/TFpRxCQS#kcSWjrM9R0i_RgazxaRtkydefFntg4L-DFOcMWlMCjU

christianhaitian, 2023-10-23:
Cool.  The experience should be the same.  Just update to the latest OTA updates using a compatible wifi adapter.