christianhaitian/arkos #813: Update OpenRCT2 in PortMaster to latest version

CavemanJoe, 2023-10-14:
2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

Yes

4. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

Yes

6. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

The version of OpenRCT2 included in Portmaster on the RG353V using the latest ArkOS suffers from this frustrating bug:
https://github.com/OpenRCT2/OpenRCT2/issues/19152
It seems this bug is fixed in later versions as of April 2023, so I would like if PortMaster used a more recent version of OpenRCT2.

### How can the issue be reproduced? (If applicable)
See reproduction details in https://github.com/OpenRCT2/OpenRCT2/issues/19152

### What device are you using?
RG353V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
I don't have the device in front of me but I updated it yesterday, so whatever version was current as of 2023-10-13

### Anything else you'd like to include that may help to review this issue or feature request?
Only to thank and congratulate you on some excellent software and ask for a donation link :)

christianhaitian, 2023-10-15:
Hi,

The original porter of OpenRCT2 for portmaster has been notified of this issue and will review when he has a chance.  You can find a donation link on the main wiki page located at the top and bottom.  http://arkos.hopto.me

closed 2023-10-15 by christianhaitian

CavemanJoe, 2023-10-15:
Thank you very much for your attention, I've put in a donation.  If you have contact details for the OpenRCT2 porter I'll go give him a donation too.