christianhaitian/arkos #814: Question about unmapped hardware button

genericapplesauce, 2023-10-15:
Hi @christianhaitian,

I know you don't officially support the r35s/rgb20s, so I'm not requesting anything, and hopefully you don't mind me asking some questions. I was just wondering if you had any hints on how I would go about getting the unused `Function` key working. On those devices there's a button in between the `Start` and `Select` keys that doesn't do anything. I took a look at the drivers that my rgb20s was loading and it seemed to register the controller as an Odroid GO. `evtest` was showing that nothing was pressed when the `Function` key was pressed. Does that imply that there's a missing driver, or is there something that needs changing in the kernel? How/where are these controller buttons usually mapped?

Thanks!

christianhaitian, 2023-10-15:
Hello,

While I don't support the rgb20s, I did add support for the r35s.  The function key doesn't currently register at all.  Either the button is placebo or the kernel driver needs to be updated to recognize it.  The kernel source for the rg351mp which the rgb20s and r35s are cloned of is here: https://github.com/christianhaitian/linux/tree/rg351

For the global hotkey driver, this is the source for it here: https://github.com/christianhaitian/ogage/tree/rg351mp
again, won't work with the function button because as we know, the button doesn't register at all although it seems to be wired in the device.

JanuszSamborski, 2024-02-04:
Hi @christianhaitian 
This could be added for `r36s`, but now it uses the same build as `rg351mp` so I'm not really sure how to handle that.
On `r36s` `FN` button does show up as this input event:
```log
Event: time 1707081272.162162, type 1 (EV_KEY), code 708 (BTN_TRIGGER_HAPPY5), value 1
Event: time 1707081272.162162, -------------- SYN_REPORT ------------
Event: time 1707081272.414091, type 1 (EV_KEY), code 708 (BTN_TRIGGER_HAPPY5), value 0
Event: time 1707081272.414091, -------------- SYN_REPORT ------------
```

Btw. is there any particular reason why ogage doesn't use config file for button mapping?

christianhaitian, 2024-02-05:
The latest 01272024-1 update enables the FN key.  For ogage not using a config file for mapping is because I don't know how to modify the code to do this.  If you or anyone else familiar with RUST programming know to to enable that, the source code for that for the rg351mp is here: https://github.com/christianhaitian/ogage/blob/rg351mp/src/main.rs