christianhaitian/arkos #815: Request: Launcher for Apple 2

danbeimborn, 2023-10-16:
Hello Christian,

Thank you so much for doing all this work, it's an amazing result!

I have a request- I've spent some time trying to get apple 2 working, but I can't seem to find a core that will launch the basic apple2p or apple2e rom. it appears the method is basically "mame apple2p -flop1 $FLOPPY_NAME.dsk". This works for me in mame, though I suspect there is something I am doing wrong when I try to create a launcher in /etc/emulationstation/es_systems.cfg.

is there a core that will currently support this, and please if so what do I need to make it work?

christianhaitian, 2023-10-16:
I've been looking for various solutions for Apple 2.  I did hear that Mame is able to do Apple 2 but not sure if it's with the standalone or libretro core.  Have you tried this with the libretro Mame core?  This is how you can run it from within a terminal session in ArkOS.

`/usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/mame_libretro.so "Your rom"`

danbeimborn, 2023-10-17:
> It seems you need a bios (cg apple2p.zip or apple2e.zip). WHen I load
> directly in mame I get the "APPLE ][" on top of screen and a cursor, but
> not with mame_libretro.so.
>
> Using the standalone mame 2003 binary, it looks like "mame apple2p -flop1
> ./PATH/TO/disk.image.dsk"
>
> The retroarch core complains about any CLI args after the apple2p.zip rom
> name.
>
> I've tried a couple approaches (with mame_libretro.so, mame2003_plus.so,
> mess.so)
>
>         <system>
>                 <name>apple2</name>
>                 <fullname>Apple 2</fullname>
>                 <path>/roms2/apple2/</path>
>                 <extension>.zip .ZIP .dsk .DSK</extension>
>                 <command>sudo perfmax %EMULATOR% %CORE%;nice -n -19
> /usr/local/bin/apple2.sh retroarch mame2003_plus %ROM% ; sudo
> perfnorm</command>
>                 <platform>arcade</platform>
>                 <theme>apple2</theme>
>         </system>
>
> [... 53 lines cut ...]
> (1986)(apple)(disk 2 of 2).po" size="143360" crc="5e9efe49"
> sha1="53d0a65fc50281fbdd9ff048ffee5012bb72061c" offset="0x0000" />
>                         </dataarea>
>                 </part>
>         </software>

OK, after experimenting more on a different ubuntu system with mame 2003
standalone binary:

*./mame apple2e -flop1 $disk_image *

apple2e.zip is BIOS / HWinfo for the system
-flop1 tells it to use the correct floppy driver size
$disk_image would be SOMENAME.dsk

The most commonly supported bios files are apple2p.zip and apple2e.zip .
Disk images seem to come in "Filename.dsk" or "Filename.DSK"

That defaults to a fullscreen apple 2 booting the disk requested. Exiting
was slightly odd, had to hit scroll lock to enable hotkeys, then esc
worked. Looks like some toggles for default full emulation (no hotkeys
work) in the config though I didn't manage to get it defaulting right.

danbeimborn, 2023-10-18:
> I just found a post online that seems like someone figured out how to run
> apple2 via retroarch.
> https://forums.launchbox-app.com/topic/70581-apple-ii-via-retroarch/

Interesting- I'll have a crack. It's a fairly complex config there, I might
actually try it out on windows first.

do you test on a "simpler" unix at all, oracle vbox or similar? I've an
rg353m + keyboard/hdmi out to get a larger screen etc, though it's still a
little easier from a PC with lots of ssh windows

christianhaitian, 2023-10-18:
I test directly in the OS.  Usually just within a ssh session.

danbeimborn, 2023-10-20:
> I test directly on device within the OS. Usually just within a ssh session.

Not sure what I'm doing wrong, but I can't seem to get retroarch to launch
an apple2 file. I've tried lots of approaches but no luck getting mame or
mess core to launch an apple2p or apple2e bios + disk image. standalone
mame on windows is pretty happy to take "mame apple2e -flop1
/path/disk_image.dsk" and launch!
I tried "apt install mame" to get standalone, though that launches with a
complaint:

arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is
'3'.Segmentation fault

I'm normally pretty low-level with unix, but even stracing the retroarch
cmds didn't show me anything obviously missing/wrong

danbeimborn, 2023-10-22:
Ok, success. I'm able to get it working and make a manual collection.
Spaces in disk image names is a problem, but mess_libretro.so handles
apple2 really nicely! We should be able to get this to now run Mac..
apple3.. applegs.. and indeed any other computer style system with
secondary disk images that MESS supports.

Here's my working es_systems.cfg snippet:
        <system>
                <name>apple2</name>
                <fullname>Apple 2</fullname>
                <path>/roms2/apple2/</path>
                <extension>.zip .ZIP</extension>
                <command>sudo perfmax %EMULATOR% %CORE%;
*/roms2/apple2/rungame.cmd* %ROM% ; sudo perfnorm</command>
                <platform>arcade</platform>
                <theme>apple2</theme>
        </system>

"apple2" already present in themes. I filled in platform "arcade", wasn't
sure what to put there!

Here is the bash script to run games that I stuck into the roms folder.. I
think it can go anywhere?

***@***.***:/roms2/apple2$ cat rungame.cmd
#!/usr/bin/bash
/usr/local/bin/retroarch -L
/home/ark/.config/retroarch/cores/mess_libretro.so "apple2e -rp
/roms2/bios/ -gameio joy -flop1 $1"

*NOTE It fails if there are spaces in the file names. *I found one chat
thread here:
https://www.arcadepunks.com/arcade-punks-arcade-building-modding-and-gaming-forum/myretrozz-classic-minis/for-those-about-to-retroarch-mame/

bios files go in /roms2/bios per the "-rp" flag in the command.. it needed
at least

20091 00:12:21.047934 openat(AT_FDCWD, "/roms2/bios/apple2e.zip", O_RDONLY
<unfinished ...>
20091 00:12:21.048753 openat(AT_FDCWD, "/roms2/bios/votrax/sc01a.bin",
O_RDONLY <unfinished ...>
20091 00:12:21.048930 openat(AT_FDCWD, "/roms2/bios/votrax.zip", O_RDONLY)
= 5
20091 00:12:21.050243 openat(AT_FDCWD, "/roms2/bios/a2diskiing.zip",
O_RDONLY) = 5
20091 00:12:21.051609 openat(AT_FDCWD, "/roms2/bios/d2fdc.zip", O_RDONLY) =
5

Note that a2diskii.zip was the bios I found from the conversation you'd
linked earlier in the thread..  I had to rename to a2diskiing.zip after
stracing to see where it failed..

To-do:

   - I haven't quite worked out controller mappings yet, that will be
   interesting. A lot of these were custom keyboard keys to map, some support
   joystick etc.
   - I think we can add state saving and config paths in the rungame.cmd cg
   -statename "$1" -state_directory "/roms2/apple2/state" -cfg_directory
   "/roms2/apple2/cfg" ?
   - Some possible hacks for romnames with spaces.. a fix script ?
   Screenscraper grabs nice thumbnails and videos for various apple2
   colletions I've found, but not if the spaces are taken out of the
   filenames. My own hack to test was to scrape them, then run a rename " " to
   "_" in both disk name/zip and video/thumbnail. Not too easy for end user..

danbeimborn, 2023-10-24:
Have been trying to find where custom controls config could go/ ideally I'd
save one with shoulder buttons mapped like

L1 Space
L2 1
R2 j
R1 enter

Left analog button = tab

that would cover most of the keypresses needed to start a game on an apple.
(space or return to cancel boot screen.. 1 for single player.. j to select
joystick).

An alternative is custom keymappings to dpad/joystick/buttons..

many games used a/z for up down, left-right arrows, space bar.
Some use I up j left k down l right
esc is often pause in game, though emulator hotkey prob better.

Also mess uses the mame core that wants you to press scroll lock first,
then tab to get the menu.. that's some kind of Mess convention for fully
emulating kbd or not I guess.. I suspect that can have a default off to
make the "Tab" work to set default controls.

Seems there are some needed analog deadzones on RG353M also- the apple
joysticks were similar (dual potentiometer) so probably need some kind of
dead zone.

do you know where the configs would go? It seems to have some attempts to
open them in /roms2/apple2/mame/cfg/ ?

christianhaitian, 2023-10-24:
Check in `/home/ark/.config/retroarch/config/`

christianhaitian, 2023-10-27:
I generally hangout mostly at the Retrogame handheld Discord.  You can find the link at the bottom of the main wiki landing page.

zomgugoff, 2023-10-28:
The script can be skipped with this:
<command>sudo perfmax %EMULATOR% %CORE%; rom=%ROM%; nice -n -19 /usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/mess_libretro.so "apple2e -rp /roms2/bios/ -gameio joy -flop1 "$rom""; sudo perfnorm</command>

Also, more tested supported filetypes:
<extension>.zip .ZIP .dsk .DSK .do .DO .po .PO .nib .NIB</extension>

zomgugoff, 2023-11-04:
Platform should be changed for scraping reasons:

<platform>apple2</platform>

zomgugoff, 2024-01-30:
I'm still using the es_systems.cfg and bash script I listed above. Using it with the current MESS core. No issues.

christianhaitian, 2024-01-30:
Ok.  I'll check it out.  Thanks

christianhaitian, 2024-04-22:
Closing this since we have Apple II emulation available now.

closed 2024-04-22 by christianhaitian