christianhaitian/arkos #816: Anbernic RG353v Wi-Fi not connecting

InscrutableOwl, 2023-10-19:
I can't seem to get ArkOS to connect to Wi-Fi no matter what I do. I've checked to see if my router has fast roaming and it is turned off. I can't get it to connect to 3 other hotspots I have. The stock OS and android have no problem. Also, after getting the "secrets left out" message it will not show any Wi-Fi points only a highlighted "%" and it won't allow for selecting any other option in the setup Wi-Fi menu. I have tried it with 2 different micro SD cards. If it helps, I am using a Mac with Sonoma

christianhaitian, 2023-10-19:
Is this a fresh install of the OS?  If so, did you at any time disable wifi the enable it again?

christianhaitian, 2023-10-19:
1.  Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3.  Once completed, do wifioff then wifion from the options>advanced section, then attempt to configure your wifi again.

If this resolves your connectivity issue, make sure to complete an update.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13037636/arkosbackup.tar.gz)

InscrutableOwl, 2023-10-19:
That backup restore did the trick. Thank you so much!

closed 2023-10-19 by InscrutableOwl