christianhaitian/arkos #817: Audio stops working after plugging usb-c hub and mini-hdmi adapter to RG353PS

cidus, 2023-10-19:
### Describe the Issue (If applicable)
There is no sound at all on my device anymore, after I plugged a miniHDMI adapter and a Usb-c hub (the hub has hdmi output that may have been detected as an audio device).
EmulationStation menu shows "SND:N/A"
In Retroarch, there are no Audio devices to choose from in Audio Section.

### How can the issue be reproduced? (If applicable)
I guess plugging one of the devices did that? I don't know which one was the culprit.

### What device are you using?
RG353PS with the RG353M image

### What version of ArkOS are you on?
2.0 10062023

### Anything else you'd like to include that may help to review this issue or feature request?
I was messing around with how it works playing on a TV. It worked but had no audio. Unplugging everything and rebooting did not restore audio.

cidus, 2023-10-19:
I tried restoring a backup of the settings I had before, it had no effect.

christianhaitian, 2023-10-19:
The RG353P/PS units are not supported in ArkOS.  You're welcome to try the instructions [here](https://github.com/christianhaitian/arkos/issues/772#issuecomment-1732564970) as a potential fix for your issue.  Good luck.

closed 2023-10-19 by christianhaitian