christianhaitian/arkos #818: Unable to choose wifi connection on rg351v

Bigpeeet, 2023-10-21:
Hi,
My wifi connection screen is frozen on my new Ambernic 351v and I'm unable to press the '+' to add a connection, nothing happens when pressing 'r1' and can't get off this screen without a hard reset.

I'm using ARKOS 2.0 with a new install. I've tried reflashing on both PC and Mac, no change. Wifi is working when using Amberelec, so pretty sure wifi hardware itself isn't the issue.

Thanks in advance!

christianhaitian, 2023-10-21:
Try this and see if it resolves the issue for you:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, Launce wifi and verify it functions correctly again

Provide an update with your results as soon as possible.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13059782/arkosbackup.tar.gz)

Bigpeeet, 2023-10-21:
Tried as instructed but am now freezing on screen 'are you sure you want to restore a backup'. I'm pressing 'A' to continue but nothing happening, also nothing when I press 'B' to exit.

File was placed in 'EASYROMS' partition, /backup folder.

Any ideas?

christianhaitian, 2023-10-21:
Hmm...it sounds like the controller is not properly recognized.  Did you have to install a different dtb file due to the screen showing a blank white screen after a fresh flash?

christianhaitian, 2023-10-21:
Also, there are a couple of other options for configuring wifi.  You can do so via retroarch by going to the settings menu then wifi.  You can also import your wifi credentials following the instructions here: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#manual-option-1

Bigpeeet, 2023-10-21:
Yes I did. Okay, I can look into importing if that's the best way

Bigpeeet, 2023-10-21:
Okay, manual wifi setup worked, thanks! 
Controller buttons working properly would also great though. I'm trying to get m8 headless running but glitchy sound and screen refresh seems to point to updating ArkOS version over wifi, which can't be accessed without the buttons working.

christianhaitian, 2023-10-21:
Do you have something plugged into the usb ports that’s not power?  If so, unplug them, reboot and see if controls work on the menus then.

Bigpeeet, 2023-10-21:
Unplugged m8 and rebooted, but no luck unfortunately. Still the same issue

christianhaitian, 2023-10-21:
can you ssh into the unit, then type `evtest` then select the `OpenSimHardware OSH` and then press the A button followed by the B button and let me know what shows up?

Bigpeeet, 2023-10-21:
Ok, here we go:

Event: time 1697874410.008332, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1697874410.008332, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1697874410.008332, -------------- SYN_REPORT ------------
Event: time 1697874410.194329, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1697874410.194329, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1697874410.194329, -------------- SYN_REPORT ------------
Event: time 1697874413.755270, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1697874413.755270, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1697874413.755270, -------------- SYN_REPORT ------------
Event: time 1697874413.888262, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1697874413.888262, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1697874413.888262, -------------- SYN_REPORT ------------

christianhaitian, 2023-10-21:
Hmmm…the buttons are different on your unit versus the unit i had previously. Try seeing if you can restore the backup file I posted earlier by pressing any other buttons.

Bigpeeet, 2023-10-21:
No response to any buttons unfortunately

christianhaitian, 2023-10-21:
While in ssh, copy and paste each of the following commands into the terminal and press enter after each one:

`sudo sed -i '/event_btn_b\="BTN_EAST"/s//event_btn_b\="BTN_SOUTH"/' /usr/local/bin/buttonmon.sh`
`sudo sed -i '/event_btn_a\="BTN_SOUTH"/s//event_btn_a\="BTN_NORTH"/'  /usr/local/bin/buttonmon.sh`

Then test the restore process again.

Bigpeeet, 2023-10-21:
Sorry Christian, no change

christianhaitian, 2023-10-21:
Then unzip and stick the .sh script in your tools folder on your roms card and then run it from the Options>Tools section.
It should restore those files so you can update the OS hopefully.

[Restore ArkOS Settings.zip](https://github.com/christianhaitian/arkos/files/13062006/Restore.ArkOS.Settings.zip)

Bigpeeet, 2023-10-22:
Tried this, but on restore, I get an error 'no arkosbackup.tar.gz file was found in the /roms/backup folder'. I can see the file in the right place, my partition is called 'easyroms' and I'm using a single SD card, if that makes any difference?

Thanks for all your help so far!

christianhaitian, 2023-10-22:
Are you sure it's in the /roms/backup folder and is named `arkosbackup.tar.gz`?

christianhaitian, 2023-10-22:
must be lowercase

Bigpeeet, 2023-10-22:
[image]

This is what I have

christianhaitian, 2023-10-22:
I found my error.  Use the script in the attached zip instead.

[Restore ArkOS Settings.zip](https://github.com/christianhaitian/arkos/files/13062162/Restore.ArkOS.Settings.zip)

Bigpeeet, 2023-10-22:
working, updating now

Bigpeeet, 2023-10-22:
Thanks again for everything. I'll let you know once updating has successfully completed

Bigpeeet, 2023-10-22:
Update completed successfully but now no 'ports' option in menu?

christianhaitian, 2023-10-22:
There needs to be a .sh file in the root ports folder.  sub folders with .sh files are no longer supported.

Bigpeeet, 2023-10-22:
Ok, thanks again Christian!

closed 2023-10-22 by christianhaitian