christianhaitian/arkos #819: Weird issues in Final Fantasy 8 suggest PS1 cores might be outdated/broken?

Metzli-Cilalti, 2023-10-21:
Hello, I'm writing to report that something in the new update might have broken the pcsx_rearmed and pcsx_rearmed_rumble cores for the PS1 emulator. I have noticed two issues that were not present in the previous updates. I'll start with the first one:

Issue Number 1: If you play the game on enhanced resolution everything goes well until you get into a battle, when you do: the HUD completely breaks (also noticing a very weird coloured swirling effect when the battles start and the screen fades like it usually does, which wasn't present at first). The names of the characters, health values, ATB bars and the text on top all blink repeatedly. This continues until I disable enhanced resolution. The very strange part about this is if I then go back to using the enhanced resolution: the blinking does NOT reappear for the rest of the battle, that is until the next battle. So basically for each battle you have to go disable enhanced resolution and then re-enable it to make the text not blink.

Issue Number 2: I initially noticed that even though I had the controller set to Dualshock and I had rumble enabled and turned on in the in-game settings the device weasn't actually vibrating, so that got me to fiddling with the settings and so I noticed that the left thumbstick input was... broken. But strangely enough this only happens in the in-game menus (in order to move the cursor up and down you have to tilt the stick left or right, while up and down actually do nothing), if you roam the map it's completely fine.

I reported these issues to the dev of pcsx_rearmed and they say that the core version I provided is actually an outdated one, which is weird considering I got it through updating from Retroarch32 in the Core Updater. So at this point I don't actually know if the issue is being caused by outdated cores, the ArkOS update, or a weird combination of both and maybe the ArkOS updated messed with the pcsx_rearmed cores somehow.

I was provided a devbuild version of the core which makes the thumbstick issue slightly more tolerable but still inaccurate, the other issues remain even with that version of the core. None of those were there in the previous versions of ArkOS (update 10062023 is when I started noticing them).

### How can the issue be reproduced?
As of now I am noticing this issue as I am playing Final Fantasy 8 on my device, I have yet to test if thse issues present themselves in other or similar forms on other games.

### What device are you using?
Currently playing on Anbernic RG353M and Steam Deck

### What version of ArkOS are you on?
I am on ARkOS 2.0 update 10162023 (latest update at the time of writing this)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-10-21:
The pcsx-rearmed core was updated 3 weeks ago to commit e34ef5a.  If you updated the cores in retroarch32 before then, try updating the cores again.

Metzli-Cilalti, 2023-10-21:
That's the commit that I am using, I told the dev as well and they, too, told me I needed to update, but if that's the last commit then I don't think I can.

christianhaitian, 2023-10-21:
I'll update it to their latest commit then.  I'll let you know when it's done.

christianhaitian, 2023-10-21:
The core is updated to the latest commit as of a few minutes ago.  Please update and let me know the status of the issues reported here.

Metzli-Cilalti, 2023-10-22:
Thanks for the prompt replies! I have tested it this moning and the rumble and thumbsticks function now do behave correctly, although the graphical bug with enhanced resolution still remains. No amount of tweaking with all the options seem to do anything other than going back and forth between standard and enhanced.

christianhaitian, 2023-10-22:
Ok.  Then that’s an issue with the emulator core itself and libretro would have to review and address that.  If the issue did not occur with a previous version of the emulator core, you could load an older version of the core into the `/home/ark/.config/retroarch32/cores` folder.  You can find previous versions of the core through the historical commits here: https://github.com/christianhaitian/retroarch-cores/commits/rg503/arm7hf/pcsx_rearmed_rumble_libretro.so.zip

You could also tey a different emulator such as swanstation or duckstation.

Good luck

closed 2023-10-22 by christianhaitian

christianhaitian, 2023-10-22:
I also forgot to include that in the (10062023) update, retroarch was updated to the 1.16.0.3 version so it could be an issue with retroarch itself which again, libretro would need to address.