christianhaitian/arkos #820: I cannot access the Wifi configuration UI nor the Update UI. No buttons are responsive, I also can not leave the UI, I only can force shutdown the device.

eimfach, 2023-10-25:
### How can the issue be reproduced? (If applicable)

Not sure. It's just that I go to Options>Wifi and nothing there is responsive. Same for Options>Update. Which means I cannot update ArkOS also, even when I plugin in ethernet cable via usb-c adapter.

### What device are you using?
Anbernic RG351V

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-10-25:
Hello,

Unzip and stick the .sh script in the link below in your tools folder on your roms card.

https://github.com/christianhaitian/arkos/files/13062162/Restore.ArkOS.Settings.zip

then

1. Take the attached compressed file (arkosbackup.tar.gz) below and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, Launch wifi and verify it functions correctly again

https://github.com/christianhaitian/arkos/files/13059782/arkosbackup.tar.gz

christianhaitian, 2023-10-27:
Hello any update on this issue?

eimfach, 2023-10-28:
Hey yeah, so I have a single SD Card Setup, I can not access the files of that SD Card on another computer (ext4 ?)
I tried to transfer the files via fat32 usb stick on the device but it doesn't mount when I try to do it via Options menu. So I don't know how to transfer files at the moment. I ran out of patience then..

christianhaitian, 2023-10-28:
The Roms partition on a single sd card should have been in exfat format and accessible via a mac or windows pc.  Oh well, take care.

closed 2023-10-28 by christianhaitian

eimfach, 2023-12-02:
I tried it again and now everything is fine.. except that the volume buttons still don't work .. mhh