christianhaitian/arkos #824: Wiki: Installation Instructions for macOS

j-rahman, 2023-10-27:
One does not need to install 3rd party software in order to "flash" images to sd-cards on macOS. Like Linux, one can use `dd`.

1. Identify your sd-card using `diskutil list`. The device id should be something like `/dev/disk5`.
2. Unmount the sd-card using `diskutil unmountDisk /dev/disk5`.
3. Flash using something like `sudo dd if=ArkOS_<Name>.img of=/dev/rdisk5 bs=4M status=progress`.

Please include this information in the Wiki. Thank you.

christianhaitian, 2023-10-27:
Good to know.  I've updated the wiki and included this as a second option for Mac users and attributed the information to you.  Thank you

closed 2023-10-27 by christianhaitian

ethielknd, 2024-09-24:
Is `rdisk5` a typo?
`sudo dd if=ArkOS_<Name>.img of=/dev/rdisk5 bs=4M status=progress` -> `sudo dd if=ArkOS_<Name>.img of=/dev/disk5 bs=4M status=progress`

christianhaitian, 2024-09-24:
It looks like it is but I don't use or have a Mac to verify. Can someone who do advise?

ethielknd, 2024-09-24:
I run the command and it works perfectly fine with `rdisk`. According to [this thread](https://superuser.com/questions/631592/why-is-dev-rdisk-about-20-times-faster-than-dev-disk-in-mac-os-x) it's because we skip the buffer and writes raw directly in the disk, which is faster.

Thanks!

christianhaitian, 2024-09-24:
Oh good to know. Thanks. 👍🏿