christianhaitian/arkos #826: Hide subfolders?

Manfred282, 2023-10-30:
Hello, is it possible to completely hide subfolders in the game list?

Example on Anbernic StockOS or EmulationStation on PC:

For PSX and Dreamcast multidisc games i have them in subfolders called ".multidisc". I just use a .m3u file and it works really well with the Anbernic StockOS or on ES on PC. The ".multidisc" folder is automaticly hidden within EmulationStation.

But with ArkOS i cant find a way to hide subfolders. The ".multidisc" folder still shows within the game list with a yellow star.

I havnt found a answer on the wiki or the closed issues so have to ask here.

Any way to completely hide subfolders?

p.s. same problem with PSP DLCs because they are in subfolders too. If i put my DLC into the "/roms/psp/ppsspp/PSP/game"
folder the PARAM.SFO files does show up on the game list.

christianhaitian, 2023-10-30:
Have you tried going into a system and press select and then set show folders to never?  Also. for psx, you can go to options>advanced section and set psx to just show .m3u files.

Manfred282, 2023-10-30:
Yes, did tried it already. If i press select and and set show folders to never then i got the game twice in the game list. So that way the example.m3u and example.chd does show up within the game list ...

sure, for psx i can use the option to just show .m3u files, but whats with other systems that use multidisc? Will it work there too? I think i could edit the es_systems.cfg and remove the extension for this systems and only left .m3u files. That should work, right?

But the main problem is still there. ArkOS is showing hidden folders (dot before the folder name) within the game list and that should not be the case.

Any chance to add a option to completly hide "hidden folders" on the game list like on the other batocera based emulationstation builds?

christianhaitian, 2023-10-30:
yes you can edit the es_systems.cfg file located in `/etc/emulationstation` folder to achieve what you want.  As for the hide hidden folders deal, I may review that in the future.  I can't gaurantee that this will be a feature added.

christianhaitian, 2023-11-04:
Hi, I was able to add this feature after reviewing Batocera's sources.  It seems to do what you're looking for.  It'd be helpful if you could confirm it functions as expected from your perspective as well.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test that hidden folders are now indeed hidden.

Your feedback would be appreciated.

Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13257953/arkosbackup.tar.gz)

Manfred282, 2023-11-05:
Hi, does work like it should now. Hidden Folders are now really hidden and normal folder with content are still displayed.

So the main problem is solved ;)

Would be nice (if you have some spare time) if you can take a look at the PSP DLC folder. Here with ArkOS its located under "/roms/psp" unlike any other cfw or ofw. DLCs für the games goes into "/roms/psp/ppsspp/PSP/game" so you have the folder "ppsspp" visible with multiple "param.pbp" files inside which doesnt load.

Is it possible to completely hide this folder within emulationstation or is there any other solution?

Tanks for you hard work :)

christianhaitian, 2023-11-06:
Try restoring the attached using the same process as outlined previously [here](https://github.com/christianhaitian/arkos/issues/826#issuecomment-1793581077)

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13266858/arkosbackup.tar.gz)

Manfred282, 2023-11-06:
I have 4 DLC folders installed and now it only shows 1 folder within emulationstation.
So it just display the ULES01392 folder and hides the last 3 remaining folders.

Take a look here:

[https://i.imgur.com/Y137xvm.png](url)
[https://i.imgur.com/1tTWA1M.png](url)

christianhaitian, 2023-11-06:
The one showing, Are you sure that folder is only within the the `/ppsspp/PSP/game` subfolder?

Manfred282, 2023-11-06:
Yes, 100% sure. Did a quick search on the EASYROMS partition and the folder exists only within "/ppsspp/PSP/game"

[https://i.imgur.com/7pqdV3H.png](https://i.imgur.com/7pqdV3H.png)

christianhaitian, 2023-11-06:
See if this updated emulationstation yields better results.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13271038/arkosbackup.tar.gz)

Manfred282, 2023-11-06:
exact the same result like before.

ULES01392 is the only folder showing so no changes from the last backup.

christianhaitian, 2023-11-06:
Sorry, I provided the previous same build.  Here's the new build.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13271273/arkosbackup.tar.gz)

Manfred282, 2023-11-06:
No problem ;)

But still the same, ULES01392 with the PARAM.PBP showing and the remaining folders hidden.

christianhaitian, 2023-11-06:
I'll see if I can recreate this issue on my end

Manfred282, 2023-11-06:
okay, i hope you get it. Good luck

christianhaitian, 2023-11-06:
Nope.  I created multiple folders within `/ppsspp/PSP/game` and placed a `PARAM.PBP` file in each and none of them show on my end.

christianhaitian, 2023-11-06:
I even named one `ULES01392` and placed a `PARAM.PBP` in it, and it doesn't show on my end.

christianhaitian, 2023-11-06:
Did you set **Parse Gameslist Only** to on in the start Advanced Settings of emulationstation?  If so, set that to off.

Manfred282, 2023-11-06:
I found the cause of the problem. Yesterday i did start the PARAM.PBP within the DLC folder which gave an error within ppsspp.

Somehow they put the PARAM.PBP into the gamelist.xml, thats why it was alway showing.

Now that i deleted the gamelist.xml the entire "ppsspp" folder is hidden.

So you're the man, outstanding, hidden folders are hidden and the ppsspp folder too, great work ;)

christianhaitian, 2023-11-06:
Awesome.  Also, that gamelist.xml should regenerate at every start of emulationstation after it scans your folders again.

christianhaitian, 2023-12-09:
This has been officially accommodated with today's update.

closed 2023-12-09 by christianhaitian