christianhaitian/arkos #828: ArkOS HDMI out working on menu, but stops when game is launched | RG353v

Certifr1ed, 2023-11-02:
2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki? yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

Finally got around to trying HDMI out on my 353M, but it seems to only work for the menu and game selection.

I plug the RG353v into HDMI out, turn it on, and see EmulationStation loading on the external display, followed by my games list. Once I select a game though (any console) HDMI goes blank, and the external display reads No Signal Found.

Could it be something in my RetroArch settings? Not too familiar with those, so haven’t touched them since I believe they should be working with the initial set up. These are my Video Output retroarch settings:

Video: gl

Monitor Index: 0 (Auto)

Video Rotation: Normal

Force-Disable sRGB FBO: Off

### What device are you using?
rg353v

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
- v2.0.10.16.2023

### Anything else you'd like to include that may help to review this issue or feature request?
no

christianhaitian, 2023-11-02:
Using my RG353M and RG353V with ArkOS (10162023), I'm able to start games via HDMI on my end.  I was able to play Pac Man Championship via Retroarch PPSSPP and the Standalone PPSSPP emulators as well as Pico-8 Celeste via the Retroarch Fake-08 emulator and the standalone pico-8 binaries.  I was able to do this on a Sony Bravia 46" LCD TV, a Vizio 43" TV and a Gateway 24" monitor.  Now, I did notice if I used a cheap mini hdmi to full size hdmi adapter, some my retroarch games would not load.  Once I used a quality hdmi cable, they loaded fine.  I've included pictures of the good hdmi cable that worked fine on my end and the problematic hdmi adapter that worked fine in the menu and some standalone emulators but retroarch would not work with.  Hopefully this helps.

[image]
[image]

closed 2023-11-02 by christianhaitian

Certifr1ed, 2023-11-02:
thanks for reply i will give that a try.