christianhaitian/arkos #829: Bluetooth game controller on rg353v device not allowing me to remap keys in retroarch

Certifr1ed, 2023-11-02:
2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki? yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

I recently purchased a anbernic device the rg353v with the arkos OS. The Bluetooth controller coneects to device and I am able to configure keys in emulation station no problem.

However once I start a game in retroarch my controls are not communicating with retroarch it would seem. I have tried mapping the controller in port controller section and no buttons are registering when pressed to do custom mapping. Correct controller is displaying in device index BTW.

Also a side note some buttons do work in game and I am able to navigate retroarch menu with dpad so I know it works somewhat but the controls are all over the place and not what they say they should be in control settings. What am I doing wrong? Please help

I have a T-3 Bluetooth controller and also tried Xbox one controller and have same issue.

### What device are you using?

rg353v

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
v2.0.10.16.2023

### Anything else you'd like to include that may help to review this issue or feature request?

i can post video if you would like? just want my bt controller to work.

christianhaitian, 2023-11-02:
Sorry, not something I can provide much help with as there are to many different types of controllers out there and I don't have them all.  I can confirm the following controller works well for me as it's what I have in my possession for testing.

https://www.amazon.com/gp/product/B0B3DH1Z4P/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

closed 2023-11-02 by christianhaitian

Certifr1ed, 2023-11-02:
Retroarch automatically mapped yours and worked?

christianhaitian, 2023-11-02:
Yes

christianhaitian, 2023-11-02:
For that particular controller

Ericthedog2, 2023-12-02:
I have this exact issue on my RG353VS. I bought the controller mentioned above and it still doesn't work correctly.