christianhaitian/arkos #831: For those that can't boot the RG351MP using ArkOS, please take the rk3326-rg351mp-linux.dtb file that's in the attached zip file and copy it to the BOOT partition of your ArkOS sd card overwriting the existing one in there.  Then see if you're able to boot into ArkOS.  I was able to boot ArkOS on the unit I have using this updated dtb file.  It disables the ability for the RG351MP to even think it's capable of 1.5Ghz as it seemed to cause the unit to not be able to boot.

YutoTV, 2023-11-06:
For those that can't boot the RG351MP using ArkOS, please take the rk3326-rg351mp-linux.dtb file that's in the attached zip file and copy it to the BOOT partition of your ArkOS sd card overwriting the existing one in there.  Then see if you're able to boot into ArkOS.  I was able to boot ArkOS on the unit I have using this updated dtb file.  It disables the ability for the RG351MP to even think it's capable of 1.5Ghz as it seemed to cause the unit to not be able to boot.

Please report whether it's successful for you or not.

Thank you,

[rk3326-rg351mp-linux.zip](https://github.com/christianhaitian/arkos/files/9079429/rk3326-rg351mp-linux.zip)

_Originally posted by @christianhaitian in https://github.com/christianhaitian/arkos/issues/515#issuecomment-1179759478_

YutoTV, 2023-11-06:
Is there a fix for this problem on the RG353V, After Install Ark OS on the SD card and plugging it in the handheld. The Boot screen showup, but it stuck after that

christianhaitian, 2023-11-06:
Someone found that by replacing the rk3566-OC.dtb file in the BOOT partition, this allows some units to boot without issue.  What it does is lower the max processor speed to 1.4Ghz vs the normal max speed of 1.8Ghz of the chip.  You can try it and report your results.

[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/13262434/rk3566-OC.zip)

YutoTV, 2023-11-06:
What about the RG353V, I only got to the ArkOS boot Screen and it black out.

christianhaitian, 2023-11-06:
The attachement I just posted may help.  Someone just posted that they were able to successfully boot their 353V with that file.

YutoTV, 2023-11-06:
Thanks, I'll check it out.

InZaineMind, 2024-02-25:
after changing it to the dtp file attached. I have a white screen only. What should i do next please?

christianhaitian, 2024-06-03:
Hello, a community member by the name of H9K posted a couple of updated dtb files that makes use of some changes done by a ROCKNIX developer (sydarn) to the voltage for the CPU.  They found that the rk3566 doesn't tolerate heat very well and that by undervolting, they were able to achieve more stable performance.  It seems to be working well for at least one user that could not boot ArkOS reliability without downclocking the CPU.  Posting it here in hopes others can help test and provided feedback.

RG353M--->[rk3566-OC.zip](https://github.com/user-attachments/files/15537492/rk3566-OC.zip)
RG353V/VS--->[rk3566-OC.zip](https://github.com/user-attachments/files/15537820/rk3566-OC.zip)

czapa86, 2024-06-03:
Tested RG353M file on the unit which is unstable on boot and unfortunately no change to me. Unit still hangs on boot :(

markmcnaughton, 2025-01-05:
> ```
>           For those that can't boot the RG351MP using ArkOS, please take the rk3326-rg351mp-linux.dtb file that's in the attached zip file and copy it to the BOOT partition of your ArkOS sd card overwriting the existing one in there.  Then see if you're able to boot into ArkOS.  I was able to boot ArkOS on the unit I have using this updated dtb file.  It disables the ability for the RG351MP to even think it's capable of 1.5Ghz as it seemed to cause the unit to not be able to boot.
> ```
> 
> Please report whether it's successful for you or not.
> 
> Thank you,
> 
> [rk3326-rg351mp-linux.zip](https://github.com/christianhaitian/arkos/files/9079429/rk3326-rg351mp-linux.zip)
> 
> _Originally posted by @christianhaitian in [#515 (comment)](https://github.com/christianhaitian/arkos/issues/515#issuecomment-1179759478)_

Tested on my RG351MP. It is an improvement. It does allow the unit to boot, particularly initially. I still get crashes on boot frequently. Games freeze often, sometimes on exit, sometimes at random. So for me this device is still unusable on the the latest Arkos, very sadly. Stock ROM works though, I used the cleaned one from Retro Game Corps. I'd really love to get Arkos working.

Is there are way (for me) to edit the DTB files? Perhaps the undervolt used on the RG353 could help?