christianhaitian/arkos #833: R35S Bluetooth options

ovid-io, 2023-11-07:
Hi,

I have a `TP-Link Archer T2UB Nano` dongle that has both wifi and bluetooth, and would like to try to use my R35S with bluetooth headphones or controller, but the options seem to be missing from RG351MP image I'm using.

How would I go about obtaining the script to test if it can work?

Thanks,
Ovid

christianhaitian, 2023-11-08:
Not available on this chipset.  That feature is only available for rk3566 devices that come with bluetooth builtin.

closed 2023-11-08 by christianhaitian

deepak-gunasekaran, 2024-10-09:
With Rocknix - I could able to connect R36S + (wifi+bluetooth) dongle and connected my bluetooth headset and can listen to music off R36S..

Shirikumo, 2024-12-06:
> With Rocknix - I could able to connect R36S + (wifi+bluetooth) dongle and connected my bluetooth headset and can listen to music off R36S..

Did you replace the default ArkOS installation on your R36S by Rocknix to achieve that ? Else, i'm curious to know how to make bluetooth work on R36S i also have wifi+bluetooth dongle...

deepak-gunasekaran, 2024-12-06:
I have prepared new sd card for arkos and tested in that