christianhaitian/arkos #836: Feature: Support for HDPacks via the Mesen core.

pikamus, 2023-11-10:
You can manually add the core via retroarch.
(would be good if it got included in arkos, would make launching the game easier)
In the core config options i've enabled HDPacks
Used the correct naming.... FTP'd into the device
added to .config / retroarch config / system /  HdPacks
it doesn't appear to load.
Ideally.. would be nice if we could put the HDPacks next to the rom file.
Mesen allows the hdpacks to be kept in zip format.
thanks.

christianhaitian, 2023-11-10:
Last I checked, the mesen core didn’t run very well on rk3326 or rk3566 devices.

pikamus, 2023-11-10:
Playing on my rg353v, seems to run well on zelda.

pikamus, 2023-11-14:
I didn't link the pack. It can be found here. They also have a 'hd pack for low end CPU's' https://www.romhacking.net/forum/index.php?topic=31559.0

pikamus, 2023-11-16:
unsure... tbh i was just excited about zelda hd.
The hd packs aren't well organised.. so it's a very niche thing. Have to dig through forums to find them.
If you are interested, here is a video compilation someone put together of different nes games with hd packs. 
https://www.youtube.com/watch?v=qYDzL5hVzSM&t=1s

christianhaitian, 2023-11-22:
I had a chance to test this some more today and the performance is bad once you load the HD Pack.  You can test for yourself by loading the HD Pack into the bios folder.  As an example, my rom file name is `Legend of Zelda, The (U) (PRG0) [!].nes`.  I created a folder and subfolder named `HdPacks/Legend of Zelda, The (U) (PRG0) [!]` within the bios folder and unzipped the low end HD pack for the game in that folder including unzipping the `ZeldaRemastered.zip` file into this subfolder, then proceeded to load the `Legend of Zelda, The (U) (PRG0) [!].nes` through the retroarch mesen core.  After about 45 seconds, the game finally loads and the audio is very crackly and the framerate is abysmal.  This will not be added.

pikamus, 2023-11-22:
Fair enough. Thanks for looking into it <3

closed 2023-11-22 by christianhaitian