christianhaitian/arkos #837: (New?) RG353V has issues with ArkOS

Aida-Enna, 2023-11-13:
### Describe the Issue (If applicable)
ArkOS fails to boot from the image most of the time. I've used a few different MicroSD cards, including an official nintendo themed switch card. I've flashed it using the recommended tool a few times. Another unit I bought for someone else booted right away with the same image and tool on their SD card, so I think it might be a hardware revision or something. I saw something about a new screen?

Once it finally gets to the "the partition will now be expanded" part, it starts partitioning and stops at "File system created successfully."

### How can the issue be reproduced? (If applicable)
It happens 9/10 times.

### What device are you using?
RG353V (Translucent Purple)

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

I cannot get to this but I'm using the latest precompiled image (10/16).

### Anything else you'd like to include that may help to review this issue or feature request?
I'm familiar with linux and programming etc so I'd be willing to try any patches/updates/etc to see if it fixes it. Maybe the new release could help? (I couldn't find a new image).

Aida-Enna, 2023-11-13:
That new DTB improved boot reliability significantly. Most times it loads into Emustation without issue. I only get black screens about every 1/5 boots. Is there a way to turn on debug mode or something and provide any information that would be useful to detect whether it should use the higher or lower clock speed based on that?

Also, will booting into ArkOS at 1.4Ghz cause all subsequent apps launched from it to run at that speed as well?

Thank you for you help and hard work!

christianhaitian, 2023-11-13:
Unfortunately no debug without getting a serial cable and soldering the points on the board.  Once the unit gets to a bad state, there's nothing you can do anymore but reboot or power down the device.  That dtb locks the max speed of the entire OS to 1.4Ghz so higher end emulation may suffer.  I suspect there's a power related issue in which some SOC bins need more voltage to run at higher speeds than others.  The chips used on these handhelds are not going to have consistent QC so what works well for one, may not work well for another.  There's been a whole discussion about this in the ArkOS firmware channel within the RetroHandhelds Discord.  You're welcome to join there and review the chat history.  The invite for RetroHandhelds Discord is available below.  You have good proof of the situation in which you brought 2 units yet one runs ArkOS fine and the other doesn't.

https://discord.gg/retrohandhelds

korikirg35xx, 2023-11-14:
I have the same issue with arkos on rg353vs . After 7-8 reboots finish first boot.Every time i turn on the device needs 3-4 reboots to load and when load arkos after a while freezes.Back to stock OS (this runs very well).

korikirg35xx, 2023-11-14:
> If you're able to see an image on screen on one of the boots, then the issue is not the screen. The issue is that some units can reliably boot ArkOS at the spec speed of 1.8Ghz and some can not. Here's a dtb you can try here to see if it gives you reliable booting. It lowers the max speed to 1.4Ghz. Just unzip it and place the rk3566-OC.dtb file in the BOOT partition overwriting the existing one. The BOOT partition is fat32 so you should be able to access it from a Mac, Windows, or Linux computer to replace that file. Let me know your results.
> 
> https://github.com/christianhaitian/arkos/files/13262434/rk3566-OC.zip

I replace the dtb file and arkos works without issues

christianhaitian, 2023-11-15:
That'll have to do for now until a better solution is found.  Perhaps I'll eventually get a unit that exhibits this issue so I can further investigate it.  For now, none of my units have this issue.

closed 2023-11-15 by christianhaitian

sammcj, 2023-12-09:
I just got a new RG353V that wouldn't boot with ArkOS - using this file fixed it for me, maybe it should be merged into the default build?

gopgnz, 2024-01-24:
> If you're able to see an image on screen on one of the boots, then the issue is not the screen. The issue is that some units can reliably boot ArkOS at the spec speed of 1.8Ghz and some can not. [...]

I would like to mention that after ripping my hair out for past 2 days this also fixed the boot up issues on my RG353V. Looks like I'm also one of the lucky owners of these (new) units.

nunits, 2024-04-16:
I had this problem in one of my two units, luckily in the second everything worked fine. Jelos or stock I had no problem... don't they ever use 1.8ghz?
As for what it could be, maybe it's just the mediocre dissipation, I'll try a mod to see if this solves it. Could you create a 1.6ghz one?

lauren7ino, 2024-04-16:
I would make an effort to swap your device.
It may be something that gets worst over time, as the summer heat arrives.
You might argue that the console is unstable on warm days. Something like that, difficult for them to reproduce.

sz-hunor, 2024-05-08:
I had the above issue on my rg353m, changed the rk3566-OC.dtb which solved the stability issues however now my right shoulder button doesn't register a long press :D, as in the button works but in a game like drill dozer for example, where you need to hold it down, it just triggers like a quick press.

Swapped the rk3566-OC.dtb file back, right shoulder button works again, crashes happen again.

Regarding the stability issues, it may well be that it's a poorly binned chip, however I don't get any stability issues in either Jelos or the default Anbernic linux the device came with (just as a point of data).

czapa86, 2024-05-12:
It is software issue. I'm on the same boat. Other OSs work without an issue. Not sure if they are using different booting method or why arkos is having this problem. Maybe arkos booting method can be changed? Hope it will be resolved somehow as its quite annoying issue. Of course arkos is the best and i don't want to change to other os.

christianhaitian, 2024-05-12:
I would agree if I could replicate the issue on any of my 7 RK3566 (RG353V v1, RG353V v2, RG353M v1, 2 RG503 units, RGB30 v1 and RGB30 v2) devices but I can not as of yet . ¯\_(ツ)_/¯

cwozny, 2024-05-17:
I know it's not a supported model, but I was having this intermittent crash/hang on my RG353PS and had swapped in the file you attached (which I believe lowers the clock rate) and haven't had the device hang on me since.

christianhaitian, 2024-05-17:
I have a device coming to me soon that has the issue.  Hopefully I will be able to resolve the issue once I can finally replicate it.

pettenao83, 2024-06-07:
> I know it's not a supported model, but I was having this intermittent crash/hang on my RG353PS and had swapped in the file you attached (which I believe lowers the clock rate) and haven't had the device hang on me since.

The same with my unit.... I also have a RG353PS and cannot run fine arkos most of the time with random crashes... with the dtb file provided it's rock stable.... I can't compare performances with previous configuration... but I think psp games needs tweaking with the new version of the file...

supermodio64, 2024-06-17:
Have a relatively old RG353V that boots to a black screen frequently. If I tap the power button to sleep/wake a few times then the menu comes up and everything works fine. To my knowledge, the issue didn’t seem to happen on a fresh install. It started when I added a bunch games. Specifically Ports, N64, DC, PSP, and Saturn as I’m testing those first. It feels like a software issue but I’ll try the file that lowers clock speed. If performance suffers as a result then I’ll have to consider JELOS again. I don’t really like JELOS because of how it tries to integrate with RetroArch which conflicts with my preferred configuration. Stock doesn’t let you update RetroArch from what I can tell. ROCKNIX makes you delete Android, no thanks. ArkOS is the most normal option I’ve found besides this damn startup issue. Hope you find a solution that doesn’t sacrifice performance. Thanks for your time and effort.

gopgnz, 2024-07-07:
> I have a device coming to me soon that has the issue. Hopefully I will be able to resolve the issue once I can finally replicate it.

Hey, just wondering, were you able to replicate it with that unit? I've been on Jelos ever since but I would still switch over to Ark if that issue was fixed.

dylmar85, 2024-07-09:
I bought an RG353V a couple weeks ago, and am having this boot issue as well. The modified DTB file allows me to boot, but with compromised performance. Hopefully you find a solution soon!

Yobiwan29, 2024-07-12:
Same here. I bought a 353m and arkos get stuck at boot. Anbernic sent me a new motherboard, but same issue with this new one... I'm really disappointed because I've worked a lot on this build and I can't get a functional second unit...

christianhaitian, 2024-07-12:
I did receive a unit that supposedly has the issue but it boots and functions consistently with my samsung and sandisk cards just fine so i still can’t replicate the issue yet.

dylmar85, 2024-07-12:
> I did receive a unit that supposedly has the issue but it boots and functions consistently with my samsung and sandisk cards just fine so i still can’t replicate the issue yet.

Does that point to an SD Card compatibility issue? 
I'm using this: 
"SanDisk 32GB Ultra microSDHC UHS-I Memory Card  - 120MB/s, C10, U1, Full HD, A1, Micro SD Card "

Yobiwan29, 2024-07-13:
> I did receive a unit that supposedly has the issue but it boots and functions consistently with my samsung and sandisk cards just fine so i still can’t replicate the issue yet.

I managed to boot with a Sandisk Ultra Plus 32GB. Fresh install. But when I swap with the Sandisk Extreme 32GB present in my first unit, it can't boot correctly. Maybe some components aren't exactly the same on the two motherboards ? It takes 2s longer to boot with Utra Plus but it's not a big deal ;)

czapa86, 2024-07-13:
So which sd card is recommend? Sandisk Ultra Plus ? I can buy one and test as my unit has problem with booting

dylmar85, 2024-07-13:
> Sandisk Ultra Plus 32GB

imma try this.. where'd you purchase yours from?

dylmar85, 2024-07-13:
> So which sd card is recommend? Sandisk Ultra Plus ? I can buy one and test as my unit has problem with booting and currently i'm using Sandisk Ultra card.

sounds like it. looks like Best Buy and Target both stock it..

czapa86, 2024-07-13:
I'm Europe based, i will look for it in big retailers or buy online

Yobiwan29, 2024-07-13:
> So which sd card is recommend? Sandisk Ultra Plus ? I can buy one and test as my unit has problem with booting and currently i'm using Sandisk Ultra card.

Yes, I tried an ultra plus 32GB and an ultra 256GB. No problem. But ! The second sd card reader doesn't recognize any of my cards :-(
I've got a RGB30 too, and it does not appreciate Sandisk Extreme serie either.

dylmar85, 2024-07-14:
went a bought a "Plus" card, and still not able to boot, but got further along in the process than other cards I've tried.

to summarize..
stock card Kioxia 16gb - hangs at ArkOS logo screen
SanDisk Ultra 32Gb - hangs at ArkOS logo screen
SanDisk Ultra PLUS 32Gb - ArkOS logo for ~10 sec, loading screen with progress bar for ~2 minutes, eventually gets to a black screen where it hangs indefinitely...

it seems like the speed and quality of the card have something to do with it, but for now the stock OS is working ok for me. I also looked into JelOS and apparently its no longer being supported.

nunits, 2024-07-15:
It has nothing to do with SD memory. Yes, the modified DTB only reduces the CPU speed to 1.4ghz and this solves the problems in all cases.
Then the problem may be related to some CPUs that are not capable of working at 1.8ghz. Although the question would be. Ambelec and Rocknix do the rk3566 work at 1.8ghz?

Galaxy13, 2024-07-18:
For me original rk3566-OC.dtb doesn't work too. With 1.4 Ghz CPU frequency it works, but if you increase frequency in dtb my device stops booting. I've tried to modify dtb file and actually came with some results. If you violate cpu policy and avoid using tcsd4525 power policy to cpu (vdd_cpu),  device actually boots, and the performance didn't suffer. From dmesg logs it seems like device can't find policy for all cpus. but booted succesfully without any problem.
DMESG logs:
`[  182.222522] rockchip-dmc dmc: cpu0 policy NULL
[  182.282531] rockchip-dmc dmc: cpu0 policy NULL
[  182.335829] rockchip-dmc dmc: cpu0 policy NULL
[  182.532567] rockchip-dmc dmc: cpu1 policy NULL
[  182.635863] rockchip-dmc dmc: cpu2 policy NULL
[  182.689148] rockchip-dmc dmc: cpu2 policy NULL
[  182.752578] rockchip-dmc dmc: cpu3 policy NULL
[  183.022583] rockchip-dmc dmc: cpu0 policy NULL
[  183.075864] rockchip-dmc dmc: cpu0 policy NULL
[  183.129231] rockchip-dmc dmc: cpu0 policy NULL
[  183.182571] rockchip-dmc dmc: cpu0 policy NULL

[  183.359161] rockchip-dmc dmc: cpu0 policy NULL
`
I attached modified dts and dtb, maybe this is not a good way to fix that problem, but for me it works flawlessly.
[rk3566-OC-fix.zip](https://github.com/user-attachments/files/16289293/rk3566-OC-fix.zip)

czapa86, 2024-07-19:
This is working, thanks! However i have no sound. I copied and renamed dtb file, should i copy dts file also? How to fix the sound?

Galaxy13, 2024-07-19:
> This is working, thanks! However i have no sound. I copied and renamed dtb file, should i copy dts file also? How to fix the sound?

Sound might be a problem with Arkos itself. Find option in advanced "fix sound" or something like that

czapa86, 2024-07-19:
Tried that without success. When i copied back orginal dtb file sound returns so it seems like problem with that new dtb unfortunately

Galaxy13, 2024-07-19:
After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz
[image]
Old fix is now obsolete and not recommended to use.
[rk3566-OC.FIXED-v2.zip](https://github.com/user-attachments/files/16317347/rk3566-OC.FIXED-v2.zip)

nunits, 2024-07-19:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS ![image](https://private-user-images.githubusercontent.com/31729476/350582926-23d12244-110b-4722-a535-3d6a4d39fc9b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE0MzA1ODksIm5iZiI6MTcyMTQzMDI4OSwicGF0aCI6Ii8zMTcyOTQ3Ni8zNTA1ODI5MjYtMjNkMTIyNDQtMTEwYi00NzIyLWE1MzUtM2Q2YTRkMzlmYzliLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE5VDIzMDQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYxYzdiNTA4YzI2YmRlODAwYjE5N2U0YTQxMzQzYmMxZWEyYzAxYTc5OWZhN2VlMzZiMDZhZmU0N2ZiYjAyZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KVEAWMVqKqtCMNkOuL8FKiJgypjxf_3B9-v4xs1PAWA) Old fix is now obsolete and not recommended to use. [rk3566-OC.FIXED-v2.zip](https://github.com/user-attachments/files/16317347/rk3566-OC.FIXED-v2.zip)

Thank you very much friend.- 1.9 is the stable limit that rk3566 can work

christianhaitian, 2024-07-20:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS ![image](https://private-user-images.githubusercontent.com/31729476/350582926-23d12244-110b-4722-a535-3d6a4d39fc9b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE0MzUxMDYsIm5iZiI6MTcyMTQzNDgwNiwicGF0aCI6Ii8zMTcyOTQ3Ni8zNTA1ODI5MjYtMjNkMTIyNDQtMTEwYi00NzIyLWE1MzUtM2Q2YTRkMzlmYzliLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIwVDAwMjAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0ZWU4N2JkODFhODczYTVmOTFkMmVkNDgwZDgyYTgxNDRhY2M3NjM2YTU1MGJiYTEwOGVkNTJlMmYzODBkODImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Lnt9509ey5AKMItCP_Hqw_4Ow5wTBecghoiSoUa3VaA) Old fix is now obsolete and not recommended to use. Use this version instead: [rk3566-OC.FIXED-v2.zip](https://github.com/user-attachments/files/16317347/rk3566-OC.FIXED-v2.zip)

What changes did you make to get this stable for your unit?

czapa86, 2024-07-20:
You are hero, sound problem which i described is caused probably because i'm using rg353m with stereo speakers. Hope your file will be adjusted for all devices

JMarques90, 2024-07-20:
Finally I was able to boot with my Sandisk SD this fix. 
Sound does not work in built in speakers, I tried with bluetooth speakers and it was ok.

Galaxy13, 2024-07-20:
> What changes did you make to get this stable for your unit?

I 've changed power regulator values. (vdd_cpu)
First, I remove <regulator-init-microvolt> value. I've found the discussion of rockchip-linux kernel developers, where they make a guess, that this part can be obsolete and unstable, and now actually there is a dts file in Torvalds repo of linux kernel for RK3566, where [this](https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3566-anbernic-rgxx3.dtsi) value isn't presented. Secondly, I've changed <compatible> line to more generic type of regulator "fcs, fan53555”. And actually that's it. In my dtb, name of regulator changed from **tcs4525@40** to **regulator@40,** but that change actually doesn't do anything. Made it, cause seems like rk3566 can have tcs4525 ot sti817 cpu power regulator.

Galaxy13, 2024-07-20:
> Finally I was able to boot with my Sandisk SD this fix. Sound does not work in built in speakers, I tried with bluetooth speakers and it was ok.

Well, my unit doesn't have that problem. I've got sound problem like yours once, but I've fix it with provided script in options -> advanced -> fix sound
You can try that. If that is not worked for you, well if you can connect to your device by ssh and provide dmesg -k logs, I'd be able to know what exactly the problem is.
And yeah, do you have 353V or 353VS unit?

czapa86, 2024-07-21:
If i provide you with orginal rg353m file would you be able to do adjustment there ? Hope this can solve problems with sound on stereo devices

ghbeta, 2024-07-21:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS [...]

with this new dtb i can boot my rg353vs finally into arkos, thank you, no sound issue too

JMarques90, 2024-07-21:
> Well, my unit doesn't have that problem. I've got sound problem like yours once, but I've fix it with provided script in options -> advanced -> fix sound You can try that. If that is not worked for [...]

My device is a RG353M, same as @czapa86 . Did not notice that is fix is only for RG353VS. Also works with 353M version but could not be for it. 
Find the attached log
[log.txt](https://github.com/user-attachments/files/16325288/log.txt)

Galaxy13, 2024-07-21:
> If i provide you with orginal rg353m file would you be able to do adjustment there ? Hope this can solve problems with sound on stereo devices

Sure thing! Give me your original dtb, I'll patch it

czapa86, 2024-07-21:
> > If i provide you with orginal rg353m file would you be able to do adjustment there ? Hope this can solve problems with sound on stereo devices
> 
> Sure thing! Give me your original dtb, I'll patch it

Much appreciated! Attaching original RG353M file.
[rk3566-OC-RG353M.zip](https://github.com/user-attachments/files/16325378/rk3566-OC-RG353M.zip)

Galaxy13, 2024-07-21:
@czapa86 @JMarques90 Here you go!
Tested on my RG353V, boots and works fine. And sound work too. The only difference I see in dtbs, is that RG353M have additional line _out-l2spk-r2hp_ in audio codec section, which enables stereo support on devices with RK817 audio chip.
Now I think it should work just fine.
[rk3566-OC-353M-FIXED.zip](https://github.com/user-attachments/files/16325693/rk3566-OC-353M-FIXED.zip)

czapa86, 2024-07-21:
YOU ARE LEGEND !!! Now indeed sound works. Thank you so much !!!

lauren7ino, 2024-07-21:
Hi guys 
I've been following this thread with interest.
Do I understand correctly that you now have a working 353M dtb with 1.9ghz?
If you need an official beta tester, please post the 353M dtb here.
Thanks

JMarques90, 2024-07-21:
Worked 😁😁😁😁😁
Finally I can use this mini console.
Thank you @Galaxy13 
Thanks to Arkos devs

dalekman9999, 2024-07-22:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS [...]

Hey, tried this, seems to be working consistent. If anything changes ill let you know. Thanks for this! :)

dalekman9999, 2024-07-24:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS [...]

so ive seen something pop up just now for me, i plugged it in to charge after turnin it on when it was low, no sound now at all. i turn up the system volume too, nothing.

Elvitomex, 2024-07-29:
Forgive my ignorance but I also have a 535m 
I open the Boot sector on my MicroSD and I can see 
rk3566-OC.dtb
rk3566-OC.dtb.bright
I downloaded the Patched version from this threath
rk3566-OC-353M-FIXED.dtb file
Should I just drop this new dtb file in the Boot folder, or rename and replace the  original dtb files for this fix to work?
regards

Galaxy13, 2024-07-29:
Hi!
Yes, just rename your _rk3566-OC.dtb_ to something like _rk3566-OC-old.dtb_, place fixed file and rename fixed file to _rk3566-OC.dtb_

dylmar85, 2024-07-31:
I went the route of going through Anbernic tech support, and they replaced my RG353V unit for free. 
Now I have no problem running ArkOS using the standard dtb file.

Copy0000, 2024-08-08:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS [...]

Thank you. Good news but I was using this and suddenly it went out. Can't you lower 1.8GHz?

Copy0000, 2024-08-09:
> After some digging, I actually find problem with CPU voltage regulator, now DMESG didn't throw any error and lscpu actually shows that now my unit can capable of 1.9 GHz in ArkOS Old fix is now [...]

thank you Good news, but my device started up smoothly but suddenly turned off while in use. I hereby report.

czapa86, 2024-08-24:
I have been testing adjusted dtb file on 353M and no issues so far. Maybe battery life is a little shorter but really not 100% sure.

supermodio64, 2024-08-24:
Sorry but my RG353V still boots to a black screen occasionally with the v2 dtb.

lauren7ino, 2024-08-25:
One thing I can't understand: The stock Linux handles both types of screens without issues. What's their trick? Even incompatible, we should at least be able to extract and implement their timings ...

dannyzam49, 2024-09-10:
> @czapa86 @JMarques90¡Aquí tienes! Probado en mi RG353V, arranca y funciona bien. Y el sonido también funciona 😄. La única diferencia que veo en dtbs es que el RG353M tiene una _salida de línea adicional l2spk-r2hp_ en la sección de códec de audio, lo que parece habilitar la compatibilidad estéreo en dispositivos con chip de audio RK817. Ahora creo que debería funcionar bien. [rk3566-OC-353M-FIXED.zip](https://github.com/user-attachments/files/16325693/rk3566-OC-353M-FIXED.zip)

Hello! First of all thank you for your great work!

I'm 1-2 days away from receiving a rg353ps and I would like to use ArkOS without problems, if it's not too much trouble could you do me the favour of modifying the .dtb file if I attach it to you so I can make the correct configuration and work at 1.8 (the maximum of series if I read correctly) and with the same modification for stereo sound of the rg353m? thanks again!

[5 more replies]