christianhaitian/arkos #839: Solarus Controls aren‘t working

Handhelder, 2023-11-16:
I can‘t get any game on Solarus get to work. They do start and run, but the controls do not take any effect. Is there any way to configure these? This even isn’t working when the Settings.dat say joypad true.

christianhaitian, 2023-11-16:
What game are you testing and what is your unit?

Handhelder, 2023-11-16:
I was testing Mystery of Solarus DX on my RG353M.

christianhaitian, 2023-11-16:
I tested Mystery of Solarus DX on my RG353M running the latest version of ArkOS which is 11042023 and it controlled fine.  You just have to use the  joystick to navigate the initial menu.  Then when in game, the dpad is used to control the Link character.  That's a limitation of the engine on this platform.

christianhaitian, 2023-11-16:
Exact link where I got the game from: https://www.solarus-games.org/games/the-legend-of-zelda-mystery-of-solarus-xd/

Handhelder, 2023-11-16:
You’re right. These are working. Thank you. In fact the only one that seems not to work is A Link to the Dream. But that’s maybe because it’s not released yet and the version from GitHub isn’t ready??

christianhaitian, 2023-11-16:
maybe.  Ok, good luck and have fun.

closed 2023-11-16 by christianhaitian