christianhaitian/arkos #840: USB DAC volume control not working properly

zomgugoff, 2023-11-18:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
N/A

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
USB DAC volume control does not function when the DAC is plugged in prior to boot

### How can the issue be reproduced? (If applicable)
-Power off the unit
-Plug in the DAC + output device
-Power unit on
-Attempt to change the volume

### What device are you using?
RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
ArkOS 2.0 (11042023)

### Anything else you'd like to include that may help to review this issue or feature request?
Weirdly, the volume control will work properly when the DAC is plugged in after booting to ES, but only in games. ES audio is still played through the speaker. This issue does not occur on the RG353M.

christianhaitian, 2023-11-18:
I can’t replicate this issue.  Volume control works fine on my usb dac with hesdphones plugged when plugged in before booting up my rgb30 unit.  The one i’m testing with is this one:  https://a.co/d/06hLeWj

christianhaitian, 2023-11-18:
Also, plugging in a usb dac after the unit is booted into the OS requires ES to be restarted for audio to route through the usb dac.  That requirement is consistent for ArkOS on all supported hardware.

zomgugoff, 2023-11-18:
Restarting ES after plugging the DAC in does indeed route audio properly and the volume control works. It's just when booting up with the DAC connected that it doesn't work properly. The volume control indicator does not appear either. That was the behavior of DACs before you updated support for them.

I'm using the Rhodium DAC. It works as expected on my RG353M.

christianhaitian, 2023-11-18:
I don't know.  I can't replicate the issue on my end.  With the usb dac plugged in while the rgb30 is off, then powering on the device, audio comes out of the usb dac just fine including ES without needing an additional restart of ES.  I just tested again using this usb dac and not only did it work as expected, volume control from the dac itself works to increase and decrease the volume as well as the middle phone button mutes and unmutes the audio.

https://a.co/d/etpsNf3

zomgugoff, 2023-11-18:
I think this might be some sort of weird timing issue, based on the fact that it works when plugged in after boot. Not a deal breaker, but maybe something for the FAQ.

christianhaitian, 2023-11-19:
Do you have quick mode enabled and are booting straight into a game and noticing this issue?  I haven't tested that.

zomgugoff, 2023-11-19:
I don't use that mode, so I had not tested it. I just did, and volume is still uncontrollable.

christianhaitian, 2023-11-20:
I'm still yet to replicate this issue.  The image used on this device is similar to the rg353m.  Just the screen specs are different for the most part.  I'll close this ticket since I can't replicate the issue on any of my units.  I'd say to make sure you're using a good sd card with your unit to make sure that's not causing an impact but I suspect you know this already.