christianhaitian/arkos #844: NES and SNES zip files won't open on Powkiddy RGB30

Mach28, 2023-11-25:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
ArkOS doesn't open zipped files in either NES or SNES games. Once a game is chosen from the file picker the screen goes black, then the ArkOS logo appears twice before dropping back to the game list. If unzipped the file will open fine, zipped it will not.

### How can the issue be reproduced? (If applicable)

### What device are you using?
Powkiddy RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
v2.0 11/4/2023

### Anything else you'd like to include that may help to review this issue or feature request?
No other issues to report. So far, all other zip files work as expected.

christianhaitian, 2023-11-25:
This most likely means something is wrong with your zip files.  If you zip files contain any other files besides the roms themselves, it can cause this issue.

closed 2023-11-25 by christianhaitian

Mach28, 2023-11-25:
Thanks for the clarification!