christianhaitian/arkos #845: Kodi on Arkos for RGB30 - inputstream.adaptive addon

milankosk, 2023-11-27:
Hello my friend. I installed Arkos on my RGB30 device, everything is working very nice except 1 thing on Kodi. There is add-on with name inputstream.adaptive which I need for some other add-ons, but there is old version of this addon and I can not install it, coz Kodi on Arkos is Nexus 20.2.0, but inputstream.adaptive add-on is for previous version 19.0 Matrix. I tried everything for reinstalling it with for Nexus, but I can not, there is always report that it is not possible. I need there that addon for YouTube addon which needs inputstream.adaptive addon. Without it I can not install YouTube addon. Just check it please or give me some advice how to do it. Thanks a lot my friend and sorry for my bad English. I wish you good day

christianhaitian, 2023-11-27:
Oh, I must not have updated that one during the update.  I'll check it out.

milankosk, 2023-11-27:
Thank you for your answer mate 👍🙂

christianhaitian, 2023-11-27:
Hi, I was able to address this addon on my end.  I was able to install the Youtube Addon and it worked well for me.  It'd be helpful if you could confirm it functions as expected from your perspective as well.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, launch Kodi and verify that you're able to install and launch a youtube video.

Your feedback would be appreciated.

Thanks,

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13481653/arkosbackup.tar.gz)

milankosk, 2023-11-27:
Oh mate, you are awesome, I will try it and then I will let you know if it
is solved. Thanks again, omw to do it now

milankosk, 2023-11-28:
Hey again. I did all what you told me, now YouTube addon is working, it is
great! Thank you, you are really master. But, I have 1 more addon where
need inputstream.adaptive addon, and that addon is still not working. Name
of addon is freeview.sk, I can send you photos from my RGB30 and I send you
zip of that addon. That addon is working on my phone or on my android box
but can not work on RGB30, I think there is some problem with
inputstream.adaptive addon. If there is some solution I will thank you
mate, if no, never mind, I am happy that you helped me with YouTube addon.
Thanks again

christianhaitian, 2023-11-28:
Is this it?  https://github.com/cache-sk/kodirepo/blob/master/repository/plugin.video.freeview.sk/plugin.video.freeview.sk-1.0.2.zip

If so, the issue is that it hasn't been updated to support Kodi 20 yet.

milankosk, 2023-11-28:
This is it:
https://github.com/cache-sk/kodirepo/tree/master/repository/repository.cache-sk
.
   When you install it on Kodi, you will see there new repository and there
you can install from it in video plugins new addon with name freewiev.sk
   After install, in freewiev.sk addon you can watch all Czech and Slovak
televisions for free. Maybe it is problem that add-on have to be update.
Anyway, thank you for all your time mate 👍🙂

christianhaitian, 2023-11-28:
I see the issue there now.  It has a problem with the inputstream.adaptive addon as you said.  I'll do some more testing to see what I can figure out.

milankosk, 2023-11-28:
Ok mate, I will wait for your result, you are the master 😀😉, and thanks

christianhaitian, 2023-11-29:
Hi, This one should work correctly.  Please test and report.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, launch Kodi and verify that you're able to install and launch the freeview.sk addon.

Again, your feedback would be appreciated.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13506438/arkosbackup.tar.gz)

christianhaitian, 2023-11-30:
I had to rebuild this addon without widevine support to fix the issue.  The is another video player (ffplay) available.  Just load your videos with the supported extensions into the movies folder and you can play them from there.  My opinion though is that Kodi is superior for these tasks though.  No donation is required but appreciated.  Take care and enjoy.  Thanks for reporting this issue and helping me test the solution.

milankosk, 2023-11-30:
No problem mate, I am happy that I can help with anything for better ARK
os, I like all your work, I think Ark os is best os for RGB30 device, main
thing why I choose Ark os is that there is Kodi in the system, coz I like
to use Kodi, it is great thing for multimedia. Thanks again for all your
time and I wish you all good in your next life 😉

christianhaitian, 2023-12-09:
This has been officially added with today's update.

closed 2023-12-09 by christianhaitian

milankosk, 2023-12-09:
Awesome, thanks for the update!

milankosk, 2023-12-09:
Hey Christian, I did update system on my RGB30 but now in my Kodi is again
problem with addon freewiev.sk
     If I open freewiev.sk addon, there is only loading addon but nothing
happens. Link is here:
 https://youtube.com/shorts/BAjZDgGva_M?si=KbVtfAJMeg47fO2P

milankosk, 2023-12-09:
Oh sorry mate, after new restart everything is working good, sorry for
previous email. Thanks for great work with update 🙂

Dňa so 9. 12. 2023, 6:42 Milan Tetanuss ***@***.***>
napísal(a):

> Hey Christian, I did update system on my RGB30 but now in my Kodi is again
> problem with addon freewiev.sk
>      If I open freewiev.sk addon, there is only loading addon but nothing
> happens. Link is here:
>  https://youtube.com/shorts/BAjZDgGva_M?si=KbVtfAJMeg47fO2P
>