christianhaitian/arkos #851: can't connect to retroachievements unless ipv6 is disabled

lourinaldi, 2023-12-07:
RGB30 running ArkOS v2.0 (11/04/2023) Update.

No idea why, but the device cannot connect to retroachievements.org unless I first disable ipv6.

Disabling ipv6 currently seems to require ssh-ing in and running a few commands via CLI.

It would be great if there were a toggle under Options to disable ipv6 permanently.

christianhaitian, 2023-12-07:
That's an issue with your network and retroacheivements.org not the OS.  I have no such issues on my end.

closed 2023-12-07 by christianhaitian

lourinaldi, 2023-12-07:
Even so, I imagine it wouldn't be super difficult to implement a simple toggle as I described. I can provide the commands I used, if need be. JelOS already has this implemented.

christianhaitian, 2023-12-07:
I may consider this in the future.  Not something I'm interested in implementing at the moment.  I'm aware how this can be achieved already.  Thanks for the offer.