christianhaitian/arkos #852: 'Update Games Lists' option returns to corrupt game list

zomgugoff, 2023-12-09:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
Yes

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes

### Describe the Issue (If applicable)
Using the new 'Update Games Lists' option results in a corrupt game list (black screen with only the theme button guide at the bottom of the screen on RGB30. The RG353M returns to what looks like the main menu, but is still on the game list). Backing out to the main menu and re-entering the game list will return the screen to normal.

### How can the issue be reproduced? (If applicable)
Open a game list
Press 'select'
Select 'Update Games Lists'
Wait for the ES progress to complete

### What device are you using?
RGB30, RG353M

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
12082023

### Anything else you'd like to include that may help to review this issue or feature request?
I'm not sure if this is a theme issue. I'm using the same them on both devices (Pixel).

christianhaitian, 2023-12-09:
When I tested with the default theme and freeplay, I didn’t experience these issues.  Please try this with those themes on your end let me know your results.

zomgugoff, 2023-12-09:
While I'm not sure what the default theme is, I've tried ArkOS-Carbon and Freeplay, both devices now result with the main menu image that is actually the game list.

christianhaitian, 2023-12-09:
Ok so the issue is related to the pixel theme

christianhaitian, 2023-12-09:
Hmm...I don't get any of these issues using the pixel theme on my rgb30 or rg353m.  I can't replicate the issue.  I did make sure to set the theme configuration to 4:3.

zomgugoff, 2023-12-09:
The black screen is related to the pixel theme, but the false main menu screen happens on both devices when set to Freeplay theme. I think it's related to a UI setting. I'll poke through what I've changed to see if I can find a culprit.

zomgugoff, 2023-12-09:
Found it. The 'Transition Style' being set to 'Fade' causes this. 'Slide' and 'Instant' do not cause it.

christianhaitian, 2024-06-10:
I believe I have a resolve for this issue.  It would be helpful if you could help me test and confirm that issue is indeed resolved.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot the device.

Verify that there is no longer an issue with the fade transition enabled after doing an update games list.  Any other testing for screen corruption after doing a Update Games lists is appreciated as well.

Let me know your results.

Thank you,

[arkosbackup.tar.gz](https://github.com/user-attachments/files/15779839/arkosbackup.tar.gz)

zomgugoff, 2024-06-10:
Confirmed: 'Fade' effect no longer is a problem. 'Slide' and 'instant' are still working properly.

christianhaitian, 2024-06-10:
Awesome.  Thanks for confirming.

closed 2024-06-28 by christianhaitian