christianhaitian/arkos #853: OGA 1.1 BE - Wifi setting does not work after updating

ommokazza, 2023-12-09:
### Describe the Issue (If applicable)
'OPTIONS > WIFI' does not work after updating in OGA 1.1 (BE)

### How can the issue be reproduced? (If applicable)
1. Download and flash arkos image to t-flash : ArkOS_RGB10_v2.0_10062023.img.xz
2. After the first boot, setup wifi and update
3. Reboot and re-run "OPTIONS > WIFI"
4. Can't run it

### What device are you using?
Odroid Go Advance 1.1 (BE)

### What version of ArkOS are you on?
Base : ArkOS_RGB10_v2.0_10062023.img.xz
After updated : 12082023

### Anything else you'd like to include that may help to review this issue or feature request?
It shows following message:
"Rga built version:version:+2017-09-28 10:12:42"

After commenting 'setfont' lines in /opt/system/Wifi.sh, it works well.

christianhaitian, 2023-12-09:
Thanks for reporting the issue.  I will address this in a future update.  For now, restoring the following backup file should fix the issue for OGA, RGB10 and RK2020 devices.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, you should be able to launch the Wifi script without issue and should resolve some other unusual text issues.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13623938/arkosbackup.tar.gz)

ommokazza, 2023-12-10:
Thanks, it works well.

christianhaitian, 2023-12-16:
I've include this with the update that was released yesterday.  Thanks again.

closed 2023-12-16 by christianhaitian