christianhaitian/arkos #858: Updated scummvm.sh to support starting ScummVM games without drilling down

EnsignRutherford, 2023-12-16:
I added code to the scummvm.sh file so that the games will start when the scummvm file is in the /roms/scummvm folder instead of the subfolder where the game is located:

fbname=$(basename "$2" .scummvm)
game=$(dirname "$2")/"$fbname"/"$(basename "$2")"
set -- "$1" "$game" "$3"

Using the game "dig" as an example:

Place the game files in /roms/scummvm/dig
Create a file called "dig.scummvm" that contains the scummvm code for "dig" that was registered in the scummvm.ini file.

When the dig.scummvm file is stored in the game folder $2 = "/roms/scummvm/dig/dig.scummvm"
When the  dig.scummvm file is stored in the /roms/scummvm folder $2 = "/roms/scummvm/dig.scummvm" but does not run the game.

The code above converts "/roms/scummvm/dig.scummvm" to "/roms/scummvm/dig/dig.scummvm" and that allows the game to run.

I'm posting this for inclusion in the build.

ER

christianhaitian, 2023-12-16:
Nice.  I'll review and if I include it, I'll modify it to account for supported systems that have a 2 sd card setup.  I will credit you with this as well if I include this.

EnsignRutherford, 2023-12-16:
Cool.  The code is generic of rom location.

fbname=$(basename "$2" .scummvm)
game=$(dirname "$2")/"$fbname"/"$(basename "$2")"
set -- "$1" "$game" "$3"

It just takes the name of the scummvm file, e.g. "dig, bbvs, "The Dig", etc. and adds that as a folder name in front of the scummvm file, so:

/roms/scummvm/dig.scummvm is converted to /roms/scummvm/dig/dig.scummvm
/roms2/scummvm/dig.scummvm is converted to /roms2/scummvm/dig/dig.scummvm

The new requirement is the game folder and the name of the scummvm file needs to be the same or else if will not work.

This is how I installed 99 ScummVM games from my other Arcade Cabinet build:
1) Copied the directories containing the games to /roms2/scummvm
2) Start ScummVM from menu.scummvm and Mass Add all the games from /roms2/scummvm
3) Run Scan_for_new_games.scummvm to generate the .scummvm files in all the folders
4) Move all the .scummvm files from the subfolders to /roms2/scummvm
5) Use Skraper to get all the artwork and create the proper gamelist.xml
6) Profit

My OCD was showing.  I wanted ES to work in that folder like the other folders without traversing into a subfolder.  :)

christianhaitian, 2023-12-27:
So where in this script do I make this modification to include this capability?

```
#!/bin/bash

directory="$(dirname "$2" | cut -d "/" -f2)"

if [[ $2 == *"Scan_for_new_games.scummvm"* ]]
then
  printf "\033c" >> /dev/tty1
  cd /${directory}/scummvm
  ./Scan_for_new_games.scummvm standalone "$2"
  printf "\n\nFinished scanning the scummvm folder for games." >> /dev/tty1
  printf "\nPlease restart emulationstaton to find the new shortcuts" >> /dev/tty1
  printf "\ncreated if any.\n" >> /dev/tty1
  sleep 5
  printf "\033c" >> /dev/tty1
elif [[ $1 == "standalone" ]] && [[ ${2,,} != *"menu.scummvm"* ]]
then
  cd /opt/scummvm
  sudo /usr/local/bin/scummvmkeydemon.py &
  DIR="$( cd "$( dirname "${2}" )" >/dev/null 2>&1 && pwd )/"
  ./scummvm --auto-detect --path="$DIR"
  sudo killall python3
  sudo systemctl restart oga_events &
elif [[ $1 == "standalone" ]] && [[ ${2,,} == *"menu.scummvm"* ]]
then
  cd /opt/scummvm
  sudo /usr/local/bin/scummvmkeydemon.py &
  ./scummvm
  sudo killall python3
  sudo systemctl restart oga_events &
else
  if  [[ ${3,,} == *"menu.scummvm"* ]]
  then
      /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so
  else
      /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so "$3"
  fi
fi
```

EnsignRutherford, 2023-12-27:
Add to the beginning of the file:

#!/bin/bash

#
# This fix should allow ScummVM games to launch from a root .scummvm file and not from a subfolder
#
# This now makes the requirement that the .scummvm file has the same name as the folder the game is stored in
# and this will work without the subfolders.
#
fbname=$(basename "$2" .scummvm)
game=$(dirname "$2")/"$fbname"/"$(basename "$2")"
set -- "$1" "$game" "$3"

Then pick up with the directory= line.

Separate note... putting my roms sdcard into my PC corrupted it so I had to recreate a 512GB card.  I just read that ArkOS supports NTFS so I'm testing that now.

EnsignRutherford, 2023-12-27:
You might want to change the scan script to place all the .scummvm files in the /roms/scummvm folder.  I wrote a linux command via ssh to move them from the folders to that folder.

christianhaitian, 2023-12-27:
> ```
> Add to the beginning of the file:
>
>
> #!/bin/bash
>
> #
> # This fix should allow ScummVM games to launch from a root .scummvm file and not from a subfolder
> #
> # This now makes the requirement that the [...]

With this change, it works to launch the game from the /roms/scummvm folder but if I move the .scummvm files to a subfolder, it doesn't launch.  Some folks like to organize their games into subfolders including launching them from there.  Any suggestions for accommodating both?

EnsignRutherford, 2023-12-27:
Let me think about it and come up with a way to make both work.

EnsignRutherford, 2023-12-27:
Ok try this:
...
#!/bin/bash

#
# This fix should allow ScummVM games to launch from a root .scummvm file and not from a subfolder
#
# This now makes the requirementt that the .scummvm file has the same name as the folder the game is stored in
# and this will work without the subfolders.  This also works if the .scummvm file is within the folder
#
fbname=$(basename "$2" .scummvm)
fbdir=$(dirname "$2")
if [ -d "$fbdir/$fbname" ]; then
   game=$(dirname "$2")/"$fbname"/"$(basename "$2")"
   set -- "$1" "$game" "$3"
fi
...

This checks to see if the .scummvm is in the same folder as the game folder or within the game folder and executes appropriately.  Tested on my build,

EnsignRutherford, 2023-12-27:
I have to go back and see what's up with Daphne games... I had Dragon's Lair running and now it hangs. :(

EnsignRutherford, 2023-12-27:
So the installation procedure would be:

1) Upload all the folders that contain the ScummVM games.  Best naming convention is the name of the game, e.g. bbvs-win, bladerunner-win, etc.
2) Run menu.scummvm and mass add all the games
3) Run the Scan_for_new_games.scummvm to create the links.
     - This is where the script should put the files into the /roms/scummvm folder or into each individual folder
4) Profit!

I use Skraper beta on the .scummvm files to get the mixart for each game.

EnsignRutherford, 2023-12-27:
Are you also responsible for PortMaster?  I'm looking into errors I'm getting trying to start avp.

EnsignRutherford, 2023-12-27:
Can I ask a question about Doom?  The Main Menu run from from doom.sh?  The right analog joystick is mapped wrong on my RG353V.  Up turns left, Down turns right, left Looks Up, right Looks down.  For all WADS.  Not sure where to fix that.

christianhaitian, 2023-12-27:
Thanks I've confirmed that the last scummvm.sh script you posted works great.

I am the initial creator of PortMaster but I've moved away from maintaining that and there's a new team managing that.  Checkout [portmaster.games](https://portmaster.games/) for more information.  You can also join the Discord for PortMaster here: https://discord.gg/m2QcSkMh

You can fix the controls in the options menu while within game or with `/home/ark/.config/gzdoom/gzdoom.ini` or `/home/ark/.config/lzdoom/lzdoom.ini` depending on which engine you're using.  The default is gzdoom.  And yea the controls work fine with lzdoom but with gzdoom the right analog stick is not configured correctly.  I never noticed that before as the controls work fine on the rg353m which I thought have the same control layout.

christianhaitian, 2023-12-27:
Hmm....I copied the gzdoom.ini from the rg353m to my rg353v and the controls seem fine now.  You can do the same by copying the `/home/ark/.config/gzdoom/gzdoom.ini.353m` to `/home/ark/.config/gzdoom/gzdoom.ini` and verify on your end.

christianhaitian, 2023-12-27:
Actually, doing a restore of the default controls for gzdoom fixes the issue as well.  Seems like the initial configuration is bad.

EnsignRutherford, 2023-12-28:
Awesome!!!  I'm in process of filling a 512GB sdcard with roms.  I had read somewhere that I thought it supported NTFS but it looks like it only supports FAT32.  Sorry if my OCD is showing.  I like things to work right. :)

christianhaitian, 2023-12-28:
NTFS is supported.  You just have to make sure the switch to sd2 was done with a ntfs card in place first.  Otherwise, do a switch to main, then insert your ntfs card, then do switch to sd2.

EnsignRutherford, 2024-01-02:
I will try that when I'm back from vacation.  I'm now using a 512GB sdcard formatted as FAT32 with no issues.  I used to install the sdcard in the PC, then back to the RG353V and back and forth and it looks like the other 512GB card I was using got corrupted because in the RG353V it showed as 99% full but when I copied all the files off it they totaled 248GB.  Turns out corruption screwed with the filesystem.  A chkdsk repair in windows created a TON of lost files so I now winscp everything over and don;t remove it.  I will format a 512GB card to NTFS and see if that works.  I thought when I tried that the ArkOS never booted up.  I'll check when I'm back from vacation, week of the 15th.

christianhaitian, 2024-01-27:
This has been added with today's update.

closed 2024-01-27 by christianhaitian