christianhaitian/arkos #859: Update server is down?

AllenDang, 2023-12-17:
I installed arkOS for rgb30 with the image in wiki, since it's rather old (Updated 10/16/2023), I try to update from option menu, but always told me OTA service is down.
And I tried the update script, still cannot connect to the server. 
So I check the update script and try to access the patch link from macOS at (http://gitcdn.link/cdn/christianhaitian/arkos/main/11042023/arkosupdate11042023.zip), it shows "Application Error".

Is the update server down?

christianhaitian, 2023-12-17:
Where are you located?

AllenDang, 2023-12-17:
China, and I also tried the China location, it's also unable to access.

AllenDang, 2023-12-17:
I can download the update zip manually, is it possible I can copy it to some folder on SD card 2 and do the update offline?

christianhaitian, 2023-12-17:
github is usually not accessible from mainland China.  Unfortunately, the updates are designed to be done via ota.  Currently, there's no solution to do them offline.

closed 2023-12-17 by christianhaitian