christianhaitian/arkos #860: Port to m17??

oleguerinsano, 2023-12-17:
Can port arkos designed for Rockchip 3326 and Rockchip 3556 to M17 with Rockchip 3126C??? or any compatible designed version for other system? (can be used on M17 and designed for any other system to M17??)

christianhaitian, 2023-12-17:
I don't know.  It's not a unit I have nor one I'm interested in.

christianhaitian, 2023-12-17:
Perhaps check with the folks at JELOS.  That system is designed to be ported on almost anything.  https://jelos.org

closed 2023-12-17 by christianhaitian

oleguerinsano, 2023-12-17:
Jelos cant have any version for Rockchip RK3126C..... If I used the arkos version for RK351M (RockChip RK3326) and want to use on M17 (Rockchip RK3126C) can I swap any files to make this version compatible when I swap many files? Thanks for your reply!

christianhaitian, 2023-12-17:
I don't know anything about the rk3126c but if it's similar to the rk3326 chip but older, you may need to copy the dtb and kernel file.  Just wild guesses.