christianhaitian/arkos #862: No sound on RG353V, boot issues.

Ongnissim, 2023-12-18:
Hello! I think that I may have another one of those *bad* RG353V devices. My friend and I each got one through Temu, and mine   works perfectly fine, while his continuously crashes / has boot issues.

I've already replaced the rk3566-OC.dtb file, and that DID improve booting, but the device still doesn't have sound.

If I reinstall the OS, then the sound works, but he'll come back to me a few days later and say that the sound has suddenly stopped. I've tried the F+L3 hotkey to no avail, changing the volume manually in the EmulationStation settings, and nothing.

In the main menu of Emulation Station it says SND: N/A. Any thoughts on how to fix this? Or any way I could help debug?

I have the device in my possession for the time being.

Thanks!

christianhaitian, 2023-12-18:
See if this fixes the issue.  Then make sure you've updated to the 12/15/2023 OTA update.

1.  Take the attached compressed file (arkosbackup.tar.gz) that's designated for your device and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3.  Once completed, reboot your device and see if sound control returns

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13708718/arkosbackup.tar.gz)

Ongnissim, 2023-12-18:
That did the trick. I kept the backup file in the folder in case it happens again. Is there any information I can get you that would help in diagnosing this?

christianhaitian, 2023-12-18:
Other than needing a unit with the issue so I can further test and hopefully resolve, not really.  Take care.

closed 2023-12-18 by christianhaitian

Ongnissim, 2023-12-18:
Thank you very much