christianhaitian/arkos #863: After the last update I can't play PC Engine CD games

jmmartinsab, 2023-12-18:
After the last update, I am unable to play PC Engine CD games that I played perfectly before this update. I don't know why the emulator disappeared. I checked the proper folder for PC Engine CD, and the games are there.

christianhaitian, 2023-12-18:
PC Engine CD show up and load just fine on my end.  Can you post a screenshot of your pcenginecd rom folder?

jmmartinsab, 2023-12-18:
this is the folder where I save the games, before it was like this and they appeared to me but since the last update I don't even see pc engine cd in the menu. 

[image]
[image]

christianhaitian, 2023-12-18:
What is inside the Akumajou folder?

jmmartinsab, 2023-12-18:
An ISO of the game

christianhaitian, 2023-12-18:
post a picture of the contents of the Akumajou folder

christianhaitian, 2023-12-19:
Any update on this?

jmmartinsab, 2023-12-19:
Sorry I was busy today, this is what was in the folder.

[image]

christianhaitian, 2023-12-19:
What version of ArkOS are you on?  You can see this information by pressing the start button then scroll down to the bottom of the screen where it says distro version.

jmmartinsab, 2023-12-19:
This version ArkOS 2.0 (12152023)

christianhaitian, 2023-12-19:
Well I can't replicate the issue on my end at all.  I sourced a copy of this game and named it similarly to how you have it named in your picture and placed it on my roms2/pcenginecd folder in a subfolder named `Akumajou Dracula X - Chi no Rondo (English v1.01)` and it shows up just fine on my end.  The only other item to try is if you have the collection hidden in the start menu somehow.  Check in the start menu>UI Settings>Visible Systems and make sure NEC - PCENGINECD is not hidden there.

jmmartinsab, 2023-12-19:
It doesn't appear in the UI options, it's as if I don't have the system, but in the previous update I had it.

[image]

christianhaitian, 2023-12-20:
If you're comfortable with ssh, you can check the `/etc/emulationstation/es_systems.cfg` file and see if pcenginecd is configured as follows in the file.  

```
	<system>
		<name>pcenginecd</name>
		<fullname>NEC - PCEngineCD</fullname>
		<path>/roms2/pcenginecd/</path>
		<extension>.pce .PCE .ccd .CCD .iso .ISO .img .IMG .chd .CHD .cue .CUE .m3u .M3U</extension>
		<command>sudo perfmax %EMULATOR% %CORE%; nice -n -19 /usr/local/bin/%EMULATOR% -L /home/ark/.config/%EMULATOR%/cores/%CORE%_libretro.so %ROM%; sudo perfnorm</command>
		   <emulators>
		      <emulator name="retroarch">
		 	<cores>
		 	  <core>mednafen_pce_fast</core>
		 	  <core>mednafen_supergrafx</core>
		 	  <core>mednafen_pce</core>
		 	</cores>
		      </emulator>
		      <emulator name="mednafen">
		 	<cores>
		 	  <core>norecord-aspect</core>
		 	  <core>norecord-full</core>
		 	  <core>record-aspect</core>
		 	  <core>record-full</core>
		 	</cores>
		      </emulator>
		   </emulators>
		<platform>pcenginecd</platform>
		<theme>pcenginecd</theme>
	</system>
```
If this is not something you're familiar or comfortable with, you can also reflash and update the os.

jmmartinsab, 2023-12-20:
It's like the code you showed. I don't know, maybe the next update will fix it.

[image]

christianhaitian, 2023-12-20:
Odd.  Try deleting the gameslist.xml and gamelist.xml.old file in the `roms2/pcenginecd` folder and see if that resolves the issue.

christianhaitian, 2023-12-22:
Closing this ticket as I can't replicate this issue on my end.  Best solution I can offer you is what I suggested above or reflashing your OS card.

closed 2023-12-22 by christianhaitian