christianhaitian/arkos #864: Quick Mode RGB30 unable to quit

droople, 2023-12-19:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

I just enabled the Quick Mode of ArkOS on my RGB30. It's convenient for playing games, but I cannot quit RetroArch. If I choose Quit RetroArch, the machine reboots into the game again.

I'm unable to enter BaRT when booting and hold B button, it's directly going to the game.

### How can the issue be reproduced? (If applicable)
Refresh install the OS, and enable the quick mode

### What device are you using?
RGB30

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
Latest 12082023

### Anything else you'd like to include that may help to review this issue or feature request?
It'll work fine if you reenable it. It's a bug. The issue is the last played script it makes when you close out random bugs for reasons unknown. Disabling quick mode deletes the problematic script, and it'll be fine when you re-enabled it.

christianhaitian, 2023-12-19:
Did you update to the latest OTA update before enabling quick mode?  There was a fix in November for some issues related to quick mode.  If you didn't run the OTA update before enabling quick mode, disabling and enabling quick mode is a logic fix.

christianhaitian, 2023-12-19:
I just test this with a fresh install then ota update to the 12152023 update and quickmode quits fine to ES without restarting the game.  So that seems fine.  As far as holding B for BaRT while in the middle of a quickmode session, yea that needs attention.  Right now, it won't work unless you exit retroarch then hold down the B button.  I'll see what I can eventually come up with to address that.

droople, 2023-12-20:
> Did you update to the latest OTA update before enabling quick mode? There was a fix in November for some issues related to quick mode. If you didn't run the OTA update before enabling quick mode, disabling and enabling quick mode is a logic fix.
Hi Chris
Thank you for the quick response

I was using 12082023 on RGB30 when I had this issue. Someone else had the same issue with an older OS. I just update my RG353M to 12152023, and no issue with quick mode, so I guess it's fixed in 12152023.

droople, 2023-12-20:
> I just tested this with a fresh install then ota update to the 12152023 update and quickmode quits fine to ES without restarting the game. So that seems fine. As far as holding B for BaRT while in the middle of a quickmode session, yea that needs attention. Right now, it won't work unless you exit retroarch then hold down the B button. I'll see what I can eventually come up with to address that.

When I had the BaRT issue, the symptom was that even holding the B button during the ArkOS logo time, I was still unable to enter BaRT.
Now, of course, it's fixed. Thanks a lot.
Cheers,

closed 2023-12-20 by christianhaitian