christianhaitian/arkos #868: ES Issue counting games available

EnsignRutherford, 2023-12-20:
Hi,

Don't think this is for ArkOS specifically.  I just noticed this issue in EmulationStation on ArkOS on the RG353V:

I have "Astrocade" folder with 27 games and the rom file.  The rom file in gamelist.xml has hidden set to true, so there's 27 games in total.  When selected in ES, "astrocade" shows "28 games available".

Same issue for the "Adventure Vision" folder where it has four games visible and the rom hidden but ES returns "5 games available". 

I'm a contributor to ES so I'm going to take a look and see if the issue is in other builds as well.

ERutherford

christianhaitian, 2023-12-20:
Interesting.  Keep this updated with what you find.

christianhaitian, 2023-12-20:
I do plan to release another update by the end of this week to add performance governor control per system and per game which involves an updated ES so if there's something fairly straightforward that can be corrected for this issue, I'd like to include that if possible.

EnsignRutherford, 2023-12-20:
I just tested this in my RetroPie build under NeoGeo adding an entry for the BIOS but hiding it and the count didn't change.  Let me debug this a bit more in the ArkOS to see if I can get the specific scenario down.

christianhaitian, 2023-12-20:
I set a rom to hidden in the cps 1 gameslist and it does still show as 1 more than actually shown in the gamelist.

EnsignRutherford, 2023-12-20:
Are you using a custom ES for ArkOS?

christianhaitian, 2023-12-20:
Yes.  https://github.com/christianhaitian/EmulationStation-fcamod/tree/503noTTS

I think I found the fix here that I can merge into the ES used here. <https://github.com/batocera-linux/batocera-emulationstation/commit/143cd4ef8c7c1b620702be001406460aad150028>

christianhaitian, 2023-12-20:
The filedata.cpp changes

EnsignRutherford, 2023-12-20:
The fix should be simple.  The count of games should not be just the count of the gameList array it has to be a search where <hidden> doesn't exist or is false.  :)

christianhaitian, 2023-12-20:
That fix from Batocera's emulationstation seems to have address the issue here as well.  You can help me test and confirm this.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1(TF1 Slot).

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot your device and test that that this issue is resolved.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13730256/arkosbackup.tar.gz)

*Note*
You'll also see a performance governor option is available for all emulators and per game.  That won't work from this restore as there's some other changes that have to be made in order for that to function correctly for you so please ignore that for now.

EnsignRutherford, 2023-12-20:
Ok I'm on it... standby!

christianhaitian, 2023-12-20:
No it won't.  Also, there's a sort by alphabetical order option in ES as well.

EnsignRutherford, 2023-12-20:
Ok. Testing.

christianhaitian, 2023-12-20:
Awesome.  Thanks for reporting and confirming this fix.

EnsignRutherford, 2023-12-20:
Also, another nit... I've been editing the es_systems.cfg as I mentioned.  I turned on alphabetical sort and I didn't like it... most notably on my arcade build I put options at the end after the last system.  I went back and changed sorting to 'no' and it kept the alphabetized sort until I restarted ES.  Sorry to nitpick.

christianhaitian, 2023-12-20:
Yep looks like it does.  I have no idea how to resolve that as that feature was added by another community member.

EnsignRutherford, 2023-12-20:
Ok no worries.  Thanks for the time and attention on a great OS for a great handheld.  My older builds used to be on RaspberryPi's built into handheld cases.  The RG353V is awesome and ArkOS makes it even better!

christianhaitian, 2023-12-20:
No problem.  I also had a few raspberry pi based handhelds.  Pigrrl 2, Pocket Pigrrl, SuperGameGirl, and last was a Kinhank RetroGamePi.  These RK3326 and RK3566 have been a much better portable handheld experience than any of those for me.

EnsignRutherford, 2023-12-20:
I worked on VirtualMan's RetroPie build, and also contributed to ES to add support for the PixelCade LED Marquee.  Love the retrogaming scene!

EnsignRutherford, 2023-12-21:
Quick question before you close this thread.  Is there a way to hide the version detail and progress bar on startup and shutdown?  I could go into the linux files and muck but I don't want to.  :)  To me it would just be a cleaner boot and shutdown.  Start already allows a user to check the version.

christianhaitian, 2023-12-21:
The version info on startup is plymouth and it just hides all the systemctl processes on the back end.  You can disable plymouth but then you’ll just see a walk of text behind it i believe.  The progress bar is part of ES.  There might be modifications to the ES code that can be made to hide that

EnsignRutherford, 2023-12-21:
Oh the progress bar I was referring to was the 4 "dots" that go from white to red.
I hide the startup text on my other linux builds for a clean startup.  I'll take a look when I find time. :)

EnsignRutherford, 2023-12-21:
Quick question... where in ArkOS does emulationstation startup?  In my build I apply --no_splash to it so it doesn't show the startup or shutdown screens.  I want to see if that works in ArkOS.

christianhaitian, 2023-12-21:
`sudo systemctl status emulationstation` runs `/usr/bin/emulationstation/emulationstation.sh`

EnsignRutherford, 2023-12-22:
Wow, --no-splash does NOT work on this version of ES.  :(  Oh well. I tried.
I tried to run emulationstation --no-splash and it's ignored.  Still shows the startup screen.

christianhaitian, 2023-12-22:
This updated emulationstation has been included with today's ota update.

closed 2023-12-22 by christianhaitian