christianhaitian/arkos #869: Save state function

mrglennjones, 2023-12-20:
hi chris, merry christmas! 

i recently updated to the latest arkos, using a r36s. 

I used to be able to hold down R3 (right analogue nub button) then press power, emulation station would save the current game state then power down the device.

this no longer seems to work in a game but functions with the game choice menu.

any ideas how to resolve this?

christianhaitian, 2023-12-20:
Sounds like you had quick mode enabled.  Enable Quick Mode from the Options>Advanced section.  If you only have Disable quick mode in that section, then do Disable quick mode, then press select do Update game list or press start>quit>restart emulationstation, then Enable Quick Mode.

mrglennjones, 2023-12-20:
Aha! It was disabled,.enabled it now that works fine! Thanks so much! 💜🕹

christianhaitian, 2023-12-20:
Awesome.  Enjoy!

closed 2023-12-20 by christianhaitian