christianhaitian/arkos #872: mednafen PSX and/or Saturn support

acjohnson, 2023-12-21:
### Describe the Issue
I see mednafen 1.29.0 is included in the current release of ArkOS. Is there any reason why it can't be used for PSX or Saturn games? I have no idea how good/bad it will run on the RK3326 (or other supported ARM processors), but I am various curious to see it work!

### What device are you using?
RG351MP clone (R6S)

### What version of ArkOS are you on?
[v2.0 (12/15/2023) Update](https://github.com/christianhaitian/arkos/releases/tag/v2.0.12.15.2023)

christianhaitian, 2023-12-21:
The mednafen psx and saturn emulators will not perform well on this platform because of no gpu acceleration with that emulator on this platform at this time.

closed 2023-12-21 by christianhaitian

acjohnson, 2023-12-21:
darn, appreciate the quick answer though. Thanks!