christianhaitian/arkos #873: ArkOS not booting

doogleballs, 2023-12-21:
Just yesterday is got a No-Name R36S portable console with ArkOS 04232022 installed ( i know its old i will update it as soon as i have a microSD reader) and some preinstalled games. 

I played on it and experienced no issues until this early afternoon, I reset the console and pressed one of the coloured buttons, (probably A or B) and was sent to a boot selection screen for different emulations programs, namely Emulation Station and Retroarch. 
I tried to get into this menu again later and it didn't work. I then reset the console to try again, and after 2 dots loaded when booting ArkOS, i was met with an infinite flashing underscore. 

I have stock SD cards which from the research i've done is not good, and after taking out the roms card ArkOS didnt boot, cycling through the four loading dots infinitely...

what do you recommend i do? from what i've gathered i should get namebrand SD cards and flash the newest version of ArkOS onto them.

christianhaitian, 2023-12-21:
What you found is the general recommendation.  There is an updated build for the r36s that community members are maintaining.  I don't maintain one for the r36s.  You can join the RetroHandheld Discord and head to the gameconsole_r36s channel for more info on this.  The invite link for the RetroHandheld Discord is https://discord.gg/retrohandhelds.

Good Luck.

closed 2023-12-21 by christianhaitian

doogleballs, 2023-12-21:
Thanks for the help.