christianhaitian/arkos #875: Parental control/playtime limitation

georgehovh, 2023-12-22:
Is there any parental control or playtime limitation?
I am on the latest version an can't find any. Basically I just want to control what my kid plays and for how log he plays it.
If there is nothing, this can be a good feature to add.
One can setup a passcode for some games and leave others without any. Also a playtime timer can be setup so after it the device can be locked/turned off.
All this can also be implemented with different users logic.

christianhaitian, 2023-12-22:
The best solution there is is manually going through each game and pressing select>edit metadata>kid game>on for games you consider kid friendly for your key then pressing the start then going to the UI settings and enabling Kid UI Mode.  Other than that, there's no other solution and none I'm interested in implementing at this time.

closed 2023-12-22 by christianhaitian

georgehovh, 2023-12-22:
Thanks, this is also a perfect solution. But what about the playtime?

christianhaitian, 2023-12-22:
Nothing for playtime