christianhaitian/arkos #876: Rg503, oga 1.1 PPSSPP emulator dissapears after update.

Tafallablas, 2023-12-23:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-12-23:
Make sure you've updated to the 12152023 ota update.  That should resolve an issue where folders with a period in it would not show.  If you're on this latest update, make sure your psp games are not in subfolders with a period in the name.

christianhaitian, 2023-12-23:
Hopefully the suggestions above solve your issues.  If not, please respond so I can troubleshoot further.

closed 2023-12-23 by christianhaitian

Tafallablas, 2023-12-23:
> Hopefully the suggestions above solve your issues. If not, please respond so I can troubleshoot further.

On my RGB20 I always update with the PSP roms in the same folder, in the last update the folder is there but ppsspp standalone has disappeared. Before this update everything worked correctly. On my rg503 the installation is clean with the SD formatted. Standalone ppsspp does not appear either. On both consoles I can access the roms with retroarch and play. The path is psp/ppsspp/psp/game.iso. I have always followed this path and it has always worked. No periods, thank you.

christianhaitian, 2023-12-23:
So your psp games don't show in the menu either?

christianhaitian, 2023-12-23:
I see the problem now.  PSP games were never supposed to to in ppsspp/psp.  They were supposed to go in th root roms/psp or roms2/psp folder per the instructions [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#playstation-portable-psp).  The ppsspp subfolder is reserved for configuration files, cheats and dlcs.  In the 12/08/2023 update, the ppsspp folder was set to not be displayed so that dlc content doesn't show in the menu.  Just move your games and related saves to the root psp folder and you should be able to play them again.

Tafallablas, 2023-12-23:
> I see the problem now. PSP games were never supposed to to in ppsspp/psp. They were supposed to go in th root roms/psp or roms2/psp folder per the instructions [...]

Ok now runs Good 👍