christianhaitian/arkos #877: RK3566 device ArkOS Emulatonstation icon and locale language related problem

ljhmd0825, 2023-12-24:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)
This phenomenon is visible in the ArkOS ES main menu of the rk3566 device.

1>
[image]

"DISPLAY SETTINGS AND INFO"
icon matching is impossible.

https://github.com/christianhaitian/EmulationStation-fcamod/blob/5af94090cf1f3d270c9dca9dfd5d7f42026587b2/es-app/src/guis/GuiMenu.cpp#L40C1-L40C1

        addEntry(_("DISPLAY SETTINGS AND INFO"), true, [this] { openDisplaySettings(); });

        addEntry(_("DISPLAY SETTINGS AND INFO"), true, [this] { openDisplaySettings(); }, "iconBrightnessctl");

I think that  "iconBrightnessctl" is added.

2>
[image]
[image]

AUDIO CARD locale language missed at one.

https://github.com/christianhaitian/EmulationStation-fcamod/blob/5af94090cf1f3d270c9dca9dfd5d7f42026587b2/es-app/src/guis/GuiMenu.cpp#L308C1-L308C1

                s->addWithLabel("AUDIO CARD", audio_card);

 maybe edited                 s->addWithLabel(_("AUDIO CARD"), audio_card);

I think that "AUDIO CARD" --> _("AUDIO CARD")

3>
[image]
[image]

AUDIO DEVICE locale language missed at one.

https://github.com/christianhaitian/EmulationStation-fcamod/blob/5af94090cf1f3d270c9dca9dfd5d7f42026587b2/es-app/src/guis/GuiMenu.cpp#L316

                auto vol_dev = std::make_shared< OptionListComponent<std::string> >(mWindow, "AUDIO DEVICE", false);

 maybe edited      auto vol_dev = std::make_shared< OptionListComponent<std::string> >(mWindow, _("AUDIO DEVICE"), false);

I think that "AUDIO DEVICE" -->  _("AUDIO DEVICE")

I am not skilled in programming and I do not know exactly which branch of the source code is responsible for these devices, but please review this issue.
I am always grateful to use great firmware.

### What device are you using?

rg351MP and OGS (via rgb10/OGA BE1.1 arkos image)

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
ArkOS 12222023
Emulationstation V2.12.0 CHMOD BUILD 20231222014205

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2023-12-24:
If you're aware of how to potentially resolve the issue, for the rg351mp use the 351v branch.  If you PR a fix, I can review it and possibly include it in a future update.

ljhmd0825, 2023-12-25:
Thanks you. This is my first time doing PR, so I'm worried about whether I made a mistake.

christianhaitian, 2023-12-26:
It seems to work fine.  Thank you and I will include it in the next update.  I've given you credit in the commit for the change and will credit you in the changelog as well for the update.

christianhaitian, 2024-01-27:
This has been addressed with today's update.

closed 2024-01-27 by christianhaitian